PX4/PX4-Autopilot #5081: Thread Blocking in Magnetometer driver

mazahner, 2016-07-16:
Hello

I'm using the hmc5883 driver in Firmware/src/drivers/hmc5883 off the latest stable branch.
It works fine on the i2c device. But, When i'm trying to set the calibration in the driver, my thread blocks:

In my application:

`  int mag_fd = open(MAG0_DEVICE_PATH,0);
  if (mag_fd <= 0) {
    printf("couldn't open %s", MAG0_DEVICE_PATH);
  } else {
    mag_scale magscale;
   /* fill magscale*/
    if (OK != ioctl(mag_fd ,MAGIOCSSCALE,(unsigned long) (&magscale))) {
      printf("failed setting calib %s", MAG0_DEVICE_PATH);;
    }
    close(mag_fd);
  }`

Thread blocks at ioctl();

I managed to make the ioctl() call work by commenting out these two lines in hmc5883.cpp:
`
int HMC5883::check_calibration()
{
	bool offset_valid = (check_offset() == OK);
	bool scale_valid  = (check_scale() == OK);

	if (_calibrated != (offset_valid && scale_valid)) {
	---->	//warnx("mag cal status changed %s%s", (scale_valid) ? "" : "scale invalid ",
	---->	//(offset_valid) ? "" : "offset invalid");
		_calibrated = (offset_valid && scale_valid);
	}

	/* return 0 if calibrated, 1 else */
	return (!_calibrated);
}
`

Any Idea why this blocks my caller thread?

Thanks for you help, 
Markus

julianoes, 2016-07-16:
Strange, so you are saying that the ioctl never returns?

Can you retest after with master which includes this fix: #5062.

bkueng, 2016-07-18:
Possibly a stack overflow. Can you look at the stack usage using `top` on the nsh console?

mazahner, 2016-07-25:
Thanks for you're replies!
from the top i can tell that it blocks the thread. (it does not segfault which would happen on a stackoverflow)
i see the specific thread that was called with the flag waiting on semaphore.

@julianoes i'll try with the mater branch.

dagar, 2016-11-08:
@mazahner any update here? Did you try master?

LorenzMeier, 2017-01-04:
This was anyway wrong on that level. Drivers should never print to the console. I fixed that in 256222d44d.

closed 2017-01-04 by LorenzMeier