PX4/PX4-Autopilot #5083: Auto-disarm in RTL reported

julianoes, 2016-07-16:
There seems to be a scary bug where RTL caused all motors to stop immediately in flight.

More details in https://groups.google.com/forum/#!topic/px4users/UxPZQ6G4KVs.

julianoes, 2016-07-16:
I had a look at the 4 logfiles:

16_27_18.px4log -> crash out of air
15_55_33.px4log -> "crash" handheld
16_41_48.px4log -> "crash" handheld
16_39_57.px4log -> all correct, handheld

I believe there are two things happening in combination which leads to the unexpected behaviour:

1. The land detector publishes "landed" which is wrong.
2. The auto disarm timeout doesn't work because the land detector only publishes when it changes meaning the check below doesn't work as expected:

```
/* Update land detector */
orb_check(land_detector_sub, &updated);
if (updated) {
	orb_copy(ORB_ID(vehicle_land_detected), land_detector_sub, &land_detector);

	if (disarm_when_landed > 0) {
		if (land_detector.landed) {
			if (_inair_last_time > 0 &&
			    (hrt_elapsed_time(&_inair_last_time) > (hrt_abstime)disarm_when_landed * 1000 * 1000)) {
				arm_disarm(false, &mavlink_log_pub, "auto disarm on land");
				_inair_last_time = 0;
			}
		} else {
			_inair_last_time = land_detector.timestamp;
		}
	}

	was_landed = land_detector.landed;
	was_falling = land_detector.freefall;
}
```

It looks like `_inair_last_time` gets only updated every time the state changes and is therefore set to the time at takeoff, so clearly more than the timeout in the past. Thus the auto_disarm would trigger immediately.

julianoes, 2016-07-16:
I tried to find a cause why the land detector would detect a landing (1.):

In POSCTL `flag_control_manual_enabled` is `true` which means the land detector [decides to in-air solely on manual input z](https://github.com/PX4/Firmware/blob/master/src/modules/land_detector/MulticopterLandDetector.cpp#L194), so throttle. 
This means the histeresis timer called `_landTimer` is not updated.

Once, we switch from POSCTL to RTL `flag_control_manual_enabled` is `false`, meaning we look at these 4 bools:

```
	if (verticalMovement || rotating || !minimalThrust || horizontalMovement) {
		// Sensed movement or thottle high, so reset the land detector.
		_landTimer = now;
		return false;
	}
```

In calm hover, all flags will be `false` except `!minimalThrust`. Unfortunately, in the crash log (16_27_18.px4log), there is one small blip where thrust is 0 right at the moment of the switch from POSCTL to RTL.

I tried to reproduce this in SITL by injecting `actuators.control[3] = 0.0f;` right at the moment of a mode switch and was able to reproduce the copter falling out of the air.

**I'm not totally convinced yet that I've found "the" case, I might be on a wrong track. However, the issues found need fixing anyway.**

tubeme, 2016-07-16:
Very very long time ago around version 1.0 I've had a series of this full stops in mid air on RTL. All the time I recovered with flipping back to manual.  I did not report it and it disappeared in later versions.

julianoes, 2016-07-17:
@tubeme: I guess you can flip to manual to catch it but only if you don't have auto disarm enabled.

AndreasAntener, 2016-07-17:
@julianoes I think your right, the `_landTimer` needs to be reset in the `manual.z` case and also the disarmed timeout cannot work the way it is now (compared to an older version of that code the problem is clear).

dagar, 2016-07-18:
Is this the same bug? http://logs.uaventure.com/view/WrfLSCwPXMvx2HwwVJGsQY
POSCTL -> LAND: landed flag set briefly

@edigotlieb FYI

[image]

julianoes, 2016-07-18:
@dagar: it's very likely if it's right after POSCTL which is a manual mode and doesn't update the hysteresis counter.

edigotlieb, 2016-07-19:
another case which happened to me [http://logs.uaventure.com/view/WrfLSCwPXMvx2HwwVJGsQY](LOG)
right at the end of the flight...

PR #5086 (2016-07-30): Fix landed and auto-disarm hysteresis
This is an attempt to fix #5083.

The issue showed two cases where hysteresis was used wrongly. Both times, the timestamp was not updated correctly or consistently. I therefore went ahead and created a class which abstracts hysteresis away and replaced the two cases.

Please review and cross-test, thanks.
closed 2016-07-30 by LorenzMeier