PX4/PX4-Autopilot #5092: Testing 1.4.1 release candidate

sirPerna, 2016-07-18:
I did some test flights on a quad with Pixhack hardware. Seems like this hardware is not the best choice as the IMO isolation tend to oscillate.  

While flying a straight line in POSCTL mode I got following issue. The copter increased altitude and yaw started turning right. When I centered the stick to stop the quad started flying backwards. From this point POSCTL wasn't working anymore for the rest of the flight. I had to switch to manual mode do safe the quad. I tried a few more times to switch back to POSCTL within the same flight but every time the quad started to fly in a random direction. 
 
http://logs.uaventure.com/view/A6PHi8W8qnW386EgEk7SPo

tumbili, 2016-07-19:
@sirPerna I had a quick look at the plots in logmuncher. The position states of the estimator diverge and are being reset several times. This mostly indicates towards a inconsistency between IMU measurements and position / height measurements. The acceleration data shows strange noise at several points in the log. I'll have a closer look when I'm home.

ecmnet, 2016-07-19:
@sirPerna Could you try for comparison the stable version 1.3.4? With latest master, I've had also strange results regarding pitch, that do not occur with 1.3.4 with the same set of parameters.

sirPerna, 2016-07-20:
I did a proper IMU vibration isolation and now it works. Maybe the old log is still interesting as other people will face such an issue too if they have high vibration on the frame. 
http://logs.uaventure.com/view/uTTPh2WZuRtDc4SJcZ5V5e#Power_PLOT

tumbili, 2016-07-21:
@sirPerna Looks much better now! I noticed that your baro innovations are rather large, it looks like the baro measurements are rather noisy.

sirPerna, 2016-07-21:
@tumbili According to Lorenz the barometer should have it's own SPI bus. This hardware runs it on a shared bus which may be the source for the additional noise.

closed 2016-08-07 by LorenzMeier