PX4/PX4-Autopilot #5102: Position Hold with vision position estimation, LPE , x-y surface drift problem

rawk-v, 2016-07-19:
Hello!
I am testing vision position estimation with PX4, LPE estimator. Vision data seems very precise, but when I switch to **position hold** mode, quadcopter begin to drift in some direction of **x-y** surface while z coordinate not change.

Software information:
PX4 v1.3.4, with LPE enabled. 

This is my flying information:
1. Log information: [http://logs.uaventure.com/view/Ub9tXZPZFDaYH787FWQnbg](http://logs.uaventure.com/view/Ub9tXZPZFDaYH787FWQnbg)
2. Position estimation from `/msf_core/pose/pose/pose/position` 
[image]
3. Position estimation from PX4, `/mavros/local_position/pose/pose/position/`:
[image]

LPE parameters
```
x   LPE_INTEGRATE [69,155] : 1
x   LPE_FLW_OFF_Z [70,156] : 0.0000
x   LPE_FLW_XY [71,157] : 0.0100
x   LPE_FLW_QMIN [72,158] : 75
x   LPE_SNR_Z [73,159] : 0.0500
x   LPE_SNR_OFF_Z [74,160] : 0.0000
x   LPE_LDR_Z [75,161] : 0.0300
x   LPE_LDR_OFF_Z [76,162] : 0.0000
x   LPE_ACC_XY [77,163] : 0.0454
x   LPE_ACC_Z [78,164] : 0.0454
x   LPE_BAR_Z [79,165] : 1.0000
x + LPE_GPS_ON [80,166] : 0
x   LPE_GPS_DELAY [81,167] : 0.2500
x   LPE_GPS_XY [82,168] : 2.0000
x + LPE_GPS_Z [83,169] : 0.0000
x   LPE_GPS_VXY [84,170] : 0.2500
x   LPE_GPS_VZ [85,171] : 0.2500
x   LPE_EPH_MAX [86,172] : 3.0000
x + LPE_VIS_XY [87,173] : 0.0000
x + LPE_VIS_Z [88,174] : 0.0000
x   LPE_VIS_ON [89,175] : 1
x   LPE_VIC_P [90,176] : 0.0500
x   LPE_PN_P [91,177] : 0.1000
x   LPE_PN_V [92,178] : 0.1000
x   LPE_PN_B [93,179] : 0.0010
x   LPE_PN_T [94,180] : 0.0010
x   LPE_FGYRO_HP [95,181] : 0.1000
x   LPE_LAT [96,182] : 40.4300
x   LPE_LON [97,183] : -86.9290
```
I think vision position estimation data is good enough, since z coordinate remain not changing. (vision data is the only source to position estimate)
So it really confuse me, could anyone help? Thanks!

mhkabir, 2016-07-19:
Estimation looks just fine. Can you post a log? Its probably bad position controller tuning.

rawk-v, 2016-07-20:
Thanks for replying, @mhkabir . This is log file [10_57_11.px4log.zip](https://github.com/PX4/Firmware/files/373171/10_57_11.px4log.zip)

rawk-v, 2016-07-20:
I am sorry @mhkabir , the previous log file is at at AltCtl mode, not PosCtl mode. I update the log file below.
And the problem should describe as: **quadcopter keep turn around in a circle** in PosCtl mode while z coordinate not change.

And this is log file [07_00_06.px4log.zip](https://github.com/PX4/Firmware/files/373621/07_00_06.px4log.zip)
I have also uploaded to [http://logs.uaventure.com/view/QcABiUm82uMMLwzAU8SjBX](http://logs.uaventure.com/view/QcABiUm82uMMLwzAU8SjBX)

It looks like this:
[image]

rawk-v, 2016-07-22:
@mhkabir any suggestions? I flied several times again, still can not solve the problem.

rawk-v, 2016-07-22:
t

rawk-v, 2016-07-22:
Mistake, sorry

mhkabir, 2016-07-22:
Are you fusing vision heading?

rawk-v, 2016-07-22:
Thanks for your help, mhkabir 
No, I didn't. Do I have to fuse vision heading?
I tried fusing vision heading before, but copter keep **rotating** in manual mode. I think that is because of my wrong setting in imu-camera orientation settings in msf framework. 
I don't really know how to set them correctly, so I turn vision heading off.
This is my msf settings.
```yaml
data_playback: false
core/core_fixed_bias: false

#########IMU PARAMETERS#######
####### pixhawk - MPU6050
core/core_noise_acc: 0.013
core/core_noise_accbias: 0.000108
core/core_noise_gyr: 0.00065
core/core_noise_gyrbias: 0.00000212

pose_sensor/pose_fixed_scale: false
pose_sensor/pose_noise_scale: 0.0
pose_sensor/pose_noise_p_wv: 0.0
pose_sensor/pose_noise_q_wv: 0.0
pose_sensor/pose_noise_q_ic: 0.0
pose_sensor/pose_noise_p_ic: 0.0
pose_sensor/pose_delay: 0.02
pose_sensor/pose_noise_meas_p: 0.005
pose_sensor/pose_noise_meas_q: 0.02
pose_sensor/pose_initial_scale: 5.60

# q_ic is the quaternion representing the rotation of the camera in IMU frame. Unit quaternion here as we rotate the coordinate frames in SVO parameters.
pose_sensor/init/q_ic/w: 1.0
pose_sensor/init/q_ic/x: 0.0
pose_sensor/init/q_ic/y: 0.0
pose_sensor/init/q_ic/z: 0.0

# p_ic is the translation between the IMU and the camera in meters.
pose_sensor/init/p_ic/x: 0.1 
pose_sensor/init/p_ic/y: 0.0
pose_sensor/init/p_ic/z: 0.065

pose_sensor/pose_absolute_measurements: true
pose_sensor/pose_use_fixed_covariance: true
pose_sensor/pose_measurement_world_sensor: false # we do not publish the world in camera frame as set in SVO parameters.

pose_sensor/pose_fixed_scale: false
pose_sensor/pose_fixed_p_ic: true
pose_sensor/pose_fixed_q_ic: true
pose_sensor/pose_fixed_p_wv: false
pose_sensor/pose_fixed_q_wv: false
```
I have a downward-facing camera with image top the same as pixhawk forward direction. So I think keep `pose_sensor/init/q_ic/w:x.0 x:0 y:0 z:0` is find.
But that can not make `ATTITUDE.roll/pitch/yaw` and `VISION_POSITION_ESTIMATE.roll/pitch/yaw` the same.

mhkabir, 2016-07-22:
Yes, that's your issue. Enable vision yaw fusion. 

Also you'll need to mess with the MSF settings to get the transforms right. Unfortunately can't help you with that.

rawk-v, 2016-07-22:
Ok, I will try to dive into MSF, thank you for your help :)

mhkabir, 2016-07-22:
Closing this for now, since not a flightstack issue. Please feel free to open a discussion on discuss.px4.io and tag me in case you need to.

closed 2016-07-22 by mhkabir

saeidkazemi, 2017-02-12:
hi
We have some unknown problems to compile your code and we would be pleased if you could help us to fix it.
I have dealt with the problem of Position Control Mode actually the system rejects this mode.
We do not have GPS and want to run the Position Control with optical Flow Sensor.
We cannot find the LPE-GPS-on in Qground to turn it off, so how can we turn off the GPS?
We set the sys_group in LPE mode but again it rejects the position control mode. Do you know the reason?

Pixhawk version 2