PX4/PX4-Autopilot #5103: load_mon stop fails and publish fails in simple test

mcharleb, 2016-07-19:
When load_mon stop is run, it repost that load_mon is not running, but it is.

make posix_sitl_default
./build_posix_sitl_default/src/firmware/posix/mainapp
px4> uorb start
px4> load_mon start
px4> load_mon stop
WARN  [load_mon] not running
px4> load_mon status
INFO  [load_mon] running

If you instrument _compute you will see that _last_idle_time is always 0 so it is never advertised or published:

`void LoadMon::_compute()
{
        if (_last_idle_time == 0) {
                /* Just get the time in the first iteration */
                _last_idle_time = 1; //system_load.tasks[0].total_runtime;
                return;
        }

        /* compute system load */
        const hrt_abstime interval_idletime = system_load.tasks[0].total_runtime - _last_idle_time;
        _last_idle_time = system_load.tasks[0].total_runtime;

        _cpuload.timestamp = hrt_absolute_time();
        _cpuload.load = 1.0f - (float)interval_idletime / (float)LOAD_MON_INTERVAL_US;

        if (_cpuload_pub == nullptr) {
                PX4_INFO("ADVERTISE");
                _cpuload_pub = orb_advertise(ORB_ID(cpuload), &_cpuload);

        } else {
                PX4_INFO("PUBLISH");
                orb_publish(ORB_ID(cpuload), _cpuload_pub, &_cpuload);
        }
}
`

mcharleb, 2016-07-19:
@julianoes FYI

julianoes, 2016-07-20:
@mcharleb good point, looks like this should only be built for NuttX.

julianoes, 2016-08-09:
Adresses in #5128.

closed 2016-08-09 by julianoes