PX4/PX4-Autopilot #5110: Drift when estimating attitude on Snapdragon eagle platform

ericye16, 2016-07-20:
Hi,
I'm noticing drift / poor attitude estimation on the Snapdragon eagle platform. For example, if I flip the drone on its back, it takes a few seconds to settle and find the correct attitude. Similarly when I flip it back, the roll angle will be wrong by up to 45 degrees and only slowly correct. What parameters can I tune to fix this? I've already tried changing parameters in Attitude Q estimator via QGroundControl but these changes don't seem to affect it.

julianoes, 2016-07-22:
If the angle is wrong by about 45 degrees I would say that the gyro scaling is off by a factor of 2. However, I can't reproduce the issues with my Snappy.

Can you try with q/inav and check the performance?

Change `posix-configs/eagle/flight/px4.config` and replace:
```
ekf2 start
```
with
```
attitude_estimator_q start
position_estimator_inav start
```

and do `make eagle_default upload`.

ericye16, 2016-07-22:
Should I be building with eagle_legacy_default or eagle_default? I've been using legacy_default this whole time...

julianoes, 2016-07-22:
You can try what works better for you. I don't really know. Legacy uses the proprietary Qualcomm drivers and default the open-source DriverFramework drivers, as far as I know.

ericye16, 2016-07-22:
Replacing the ekf2 estimator with the attitude_estimator_q and position_estimator_inav does fix it. If it were the gyro scaling problem, where should I look to start debugging it? in the ekf2 module or lower (like at the drivers?)

julianoes, 2016-07-22:
If it were a gyro scaling issue, it would be in the driver.

ericye16, 2016-07-22:
Wait, wouldn't a gyro scaling issue in the driver also affect attitude_estimator_q or position_estimator_inav as well?

julianoes, 2016-07-22:
Correct, that's why I'm assuming there is no problem. If the ekf2 doesn't behave, there might be another issue there (maybe related to Snapdragon). I'll do more testing/test flying soon.

ericye16, 2016-07-22:
Here are some gifs to help you out.
![Uploading VID_20160722_113351.gif…](attitude_estimator_q and position_estimator_inav
[image]
)

I know the quality is terrible but hopefully you can see that the horizon "lags" in the ekf2 video.

mhkabir, 2016-07-23:
That version of QGroundcontrol is prehistoric!

julianoes, 2016-07-23:
@ericye16 Thanks for the videos, that's very helpful. I just want to try to understand which drivers exactly you are running.

Can you please share the output of:

```
adb shell "cat /home/linaro/mainapp.config"
adb shell "cat /usr/share/data/adsp/px4.config"
```

julianoes, 2016-07-23:
@ericye16 nevermind, I can reproduce your issue. Looking into it.

julianoes, 2016-07-23:
@ericye16 I have a potential fix.

Can you cross-test https://github.com/PX4/DriverFramework/pull/132 please?

ericye16, 2016-07-24:
> Can you cross-test PX4/DriverFramework#132

Sure, I'll test it out tomorrow (Monday) at the office.

ericye16, 2016-07-25:
Checked it out and it works great now! Thanks a bunch. :shipit:

julianoes, 2016-07-25:
Cool, I'll get it to master tomorrow then.

julianoes, 2016-07-26:
https://github.com/PX4/DriverFramework/pull/132 is merged, closing.

closed 2016-07-26 by julianoes