PX4/PX4-Autopilot #5114: Support for reposition "ground speed"

superware, 2016-07-21:
Following #4347, MAV_CMD_DO_REPOSITION "ground speed" parameter should determine the target ground speed.

For example, this will be very useful when in loiter-auto and navigating to a new position (in QGC, click on map and slide to confirm) in a speed different than the param one. It doesn't make sense to only have one predefined speed.

In addition, preparing a mission only for a single/simple navigation is very cumbersome.

([Relevant chat](https://gitter.im/PX4/Firmware?at=578dd4d39f35137e67d42d93))

dagar, 2016-07-22:
On the PX4 side adding speed to reposition is straightforward, but we need QGC to send it. Can you open a corresponding issue there?

dagar, 2017-04-28:
@superware I can implement this if you're still interested and able to test.

superware, 2017-04-30:
Sure, that will be great. I'll test it once implemented.

I guess MAV_CMD_DO_CHANGE_SPEED should also change current "target speed". Will it be possible to receive current "target speed" through MAVLink (reported by PX4)?

Thanks.

dagar, 2017-04-30:
You can get the current x,y,z velocity setpoint in POSITION_TARGET_LOCAL_NED (http://mavlink.org/messages/common#POSITION_TARGET_LOCAL_NED).

closed 2018-02-21 by PX4BuildBot