PX4/PX4-Autopilot #5115: EKF2 baro innovation varinance jumps

tumbili, 2016-07-21:
@priseborough @CarlOlsson I noticed in a recent log that the barro innovation variance looks fishy.
It seems to jump between two regimes. I'm not aware of the latest changes that went into the estimator but it could also be a logging issue.
I'll look at this as soon as possible but maybe you already have an idea.

[image]

The log is here http://logs.uaventure.com/view/uTTPh2WZuRtDc4SJcZ5V5e#Power_PLOT

CarlOlsson, 2016-07-21:
We are currently using version 16bdbebdc6c97bbd58bd9de1f12791ecc5421e3a of ecl without this issue.

[image]

However we will update to newest one in a couple of days and will report back then.

CarlOlsson, 2016-08-03:
Yes we also have the issue now after the latest subtree update:
**Before subtree update**
[image]
[image]
***After subtree update*
[image]
[image]

Green lines are forward transition, red lines backwards transition. We switch GPS module when transitioning but this should of course not influence the mag so the PzI looks really strange after subtree update

priseborough, 2016-08-03:
This needs to be checked using the replay facility to determine if it is a reporting bug or something more serious. Can someone please provide a replay log without dropouts.

CarlOlsson, 2016-08-03:
I will upload one here tomorrow

tumbili, 2016-08-04:
@CarlOlsson Did you manage to record a log?

CarlOlsson, 2016-08-05:
Not yesterday in flight because of [several strange stuff](https://github.com/PX4/Firmware/issues/5245) going on, but I have been looking into it now. I just recorded three logs on the bench with standard logging and the issue is present every time, however I also did three logs with RPL logging and the issue is gone so it seems like a logging bug.

CarlOlsson, 2016-08-05:
And it seems to be related to GPS.
Here is a plot from a standard logging with the airplane standing on the bench and gps turned off (*gps stop* in nutshell)
[image]
Here is a plot of the same thing but with fake gps activated
[image]

tumbili, 2016-08-05:
@CarlOlsson That's a good hint

tumbili, 2016-08-05:
@CarlOlsson Found the issue, I'm going to make a PR.

CarlOlsson, 2016-08-05:
sweet(Y)

tumbili, 2016-08-05:
Addressed in https://github.com/PX4/ecl/pull/180

closed 2016-08-05 by tumbili