PX4/PX4-Autopilot #5118: Mixer loading corner case

AndreasAntener, 2016-07-21:
If we try to load the following mixer it seems to get sent to IO but one of the last 2 mixers doesn't get loaded.

```
R: 2- 10000 10000 10000 0
Z:
M: 2
O:      10000  10000      0 -10000  10000
S: 1 0  10000  10000      0 -10000  10000
S: 1 1  10000  10000      0 -10000  10000
M: 2
O:       10000   10000    0 -10000  10000
S: 1 0   10000   10000    0 -10000  10000
S: 1 1  -10000  -10000    0 -10000  10000
```

It seems to split it up at a bad location during transfer. If I extend it a bit (add another null mixer before), or remove one of the minus signs in the first servo mixer, or change `PX4IO_P_CONFIG_MAX_TRANSFER` from 64 to 32, the mixer loads correctly.

LorenzMeier, 2016-12-28:
Is this still an issue?

AndreasAntener, 2016-12-28:
Are you suggesting I should try on master again? :)

LorenzMeier, 2016-12-28:
Yeah, that would be helpful. If its not fixed I'm going to try to build a test case and fix it. I'm not going to tolerate any issues any more 8)

AndreasAntener, 2016-12-28:
Still the same. Pasting the above mixer in any of the configs and loading it results in only 4 active outputs.

```
device: /dev/pwm_output0
channel 1: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1080 us, max: 1950 us, trim:  0.00)
channel 2: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1080 us, max: 1950 us, trim:  0.00)
channel 3: 1515 us (alternative rate: 400 Hz failsafe: 1500, disarmed: 900 us, min: 1080 us, max: 1950 us, trim:  0.00)
channel 4: 1515 us (alternative rate: 400 Hz failsafe: 1500, disarmed: 900 us, min: 1080 us, max: 1950 us, trim:  0.00)
channel 5: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 6: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 7: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 8: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel group 0: channels 1 2
channel group 1: channels 5 6 7 8
channel group 2: channels 3 4
nsh> WARN  [commander] Not ready to fly: Sensors not set up correctly
cat /etc/mixers/vtol_quad_x.main.mix
R: 2- 10000 10000 10000 0
Z:
M: 2
O:      10000  10000      0 -10000  10000
S: 1 0  10000  10000      0 -10000  10000
S: 1 1  10000  10000      0 -10000  10000
M: 2
O:       10000   10000    0 -10000  10000
S: 1 0   10000   10000    0 -10000  10000
S: 1 1  -10000  -10000    0 -10000  10000
```

Doing one of the described hacks then results in a working config:

```
device: /dev/pwm_output0
channel 1: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1080 us, max: 1950 us, trim:  0.00)
channel 2: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1080 us, max: 1950 us, trim:  0.00)
channel 3: 1515 us (alternative rate: 400 Hz failsafe: 1500, disarmed: 900 us, min: 1080 us, max: 1950 us, trim:  0.00)
channel 4: 1950 us (alternative rate: 400 Hz failsafe: 2100, disarmed: 900 us, min: 1080 us, max: 1950 us, trim:  0.00)
channel 5: 1500 us (default rate: 50 Hz failsafe: 1500, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 6: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 7: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 8: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel group 0: channels 1 2
channel group 1: channels 5 6 7 8
channel group 2: channels 3 4
nsh> cat /etc/mixers/vtol_quad_x.main.mix
R: 2- 10000 10000 10000 0
Z:
M: 2
O:      10000  10000      0 -10000  10000
S: 1 0  10000  10000      0 10000  10000
S: 1 1  10000  10000      0 -10000  10000
M: 2
O:       10000   10000    0 -10000  10000
S: 1 0   10000   10000    0 -10000  10000
S: 1 1  -10000  -10000    0 -10000  10000
```

LorenzMeier, 2017-01-01:
Unit test to reproduce here: https://github.com/PX4/Firmware/pull/6226

AndreasAntener, 2017-01-04:
Fixed via #6245