PX4/PX4-Autopilot #5122: Support MAV_CMD_DO_CHANGE_SPEED in auto-loiter

superware, 2016-07-22:
It is much desired to set a general dynamic "target speed" for auto-loiter mode. Currently there is only one predefined speed as a param, this should act as the default (and obviously maximum) speed, and a COMMAND_LONG with MAV_CMD_DO_CHANGE_SPEED should set the "current speed" for following MAV_CMD_DO_REPOSITION commands.

General use case: clicking on the map at a position 1.5 km away, confirming re-position, PX4 starts navigating to the new position, the pilot decides to decrease speed for a section of 500 m in order to operate and point the gimbaled camera, the pilot decides to increase speed in order to reach destination as fast as possible.

superware, 2017-12-06:
Hi @dagar, do you know if this is implemented? thanks.

dagar, 2017-12-06:
No, but it's a fairly easy addition. My only concern is how a DO_CHANGE_SPEED is respected. If you set it during mission should it be preserved if you switch to loiter? I could see people wanting that most of the time, but from a safety perspective I think any mode change should reset to the default.

superware, 2017-12-07:
Think about altitude, what's the difference? IMHO the pilot should be aware of his current speed.

There's "max speed" as a param (which makes sense since it's a platform "configuration" value), and now we're talking about "current max target speed", right?

dagar, 2017-12-07:
Altitude is unambiguous because it's in the actual waypoint. DO_CHANGE_SPEED is a command that can be inserted at any point or sent as a separately as a command. 

The question is what does that DO_CHANGE_SPEED apply to? The vehicles commanded speed for the entire mission, particular mission item or section, or even entire session?

It seems clear to me that DO_CHANGE_SPEED should be a temporary adjustment in some sense. Otherwise why not change the trim/cruise speed parameter directly?

hamishwillee, 2017-12-07:
FWIW I think that it should reset between mode changes but persist through the mission.

superware, 2017-12-19:
> Otherwise why not change the trim/cruise speed parameter directly?

Does it make sense to change a flash parameter in order to temporarily set current speed? I think parameters should be kept for configuration (such as max speed etc).

Why not have a single global "current speed"? if the pilot is entering a specific mission it is his responsibility to check and verify all relevant information including waypoints positions, altitudes and speeds.

This "current speed" should never exceed the configured "max speed" parameter anyway.

superware, 2018-01-04:
@dagar :)

dagar, 2018-01-08:
Are we in agreement then? DO_CHANGE_SPEED is preserved for the duration of the mission?

hamishwillee, 2018-01-08:
@dagar Depends on whether you're saying it doesn't change even on mode change or not.

superware, 2018-01-08:
It makes sense to have a global target_speed, which can be changed at any mode.

When entering mission mode, target_speed should be kept aside, the mission might change it via DO_CHANGE_SPEED or MAV_CMD_DO_REPOSITION, and when exiting mission - the previously kept value should be the new target_speed.

hamishwillee, 2018-01-08:
@superware Indeed, hence my request for clarification. I don't really mind what we do but I clearly need to document it. 
At the moment my understanding is that if you are in a mission and you set the DO_CHANGE_SPEED this speed will be used through out the rest of the mission. So if you have a loiter mission item you'll circle at that speed in FW. 
If however you were to change the mode to HOLD, and then go back to mission would the mission keep its currently set DO_CHANGE_SPEED or would that have been reset by going into hold mode?

superware, 2018-01-09:
Well, I think it's right to always keep the inner-mission target_speed when resuming a mission. Sending an explicit DO_CHANGE_SPEED (via COMMAND_LONG etc) while in mission should set that inner-mission target_speed, obviously.

hamishwillee, 2018-01-09:
Sounds good to me. As I said, I just want a clear description :-)

superware, 2018-01-15:
Hi @dagar, what do you think?

dagar, 2018-02-12:
Sorry, I haven't had time to implement this. @superware If you're interested I could help point you in the right direction to make the change.

superware, 2018-02-12:
@dagar I have no experience in this, do you think it will be easy/safe to implement the most basic logic?

superware, 2019-08-25:
@dagar hey, can you please reopen?

superware, 2019-09-24:
Besides MAV_CMD_DO_CHANGE_SPEED which is still missing, I've tried utilizing MAV_CMD_DO_REPOSITION Speed param.

It seems to work well, except that it's not possible to ONLY set the speed (with all other params as NaN) since NaN X/Y/Z is actually "hold in place". Keep in mind that a specific Z but NaN X/Y maintains the last X/Y set-point, so NaN sometime means "don't change" and sometimes "change to current".

My suggestion is to use the Bitmask param (MAV_DO_REPOSITION_FLAGS) to control the different settings of each reposition command.

BazookaJoe1900, 2020-02-21:
still relevant

superware, 2020-05-22:
Still relevant.

BazookaJoe1900, 2020-11-23:
Still relevant