PX4/PX4-Autopilot #5130: Mavlink mission sync + MissionFeasibilityChecker

dagar, 2016-07-22:
I was looking at moving MissionFeasibilityChecker to a lib so that it could also be used during mavlink mission sync. There are a few things that would have to change (the way it handles geofence), but it looks doable.

Any comments/suggestions/objections before I try?

julianoes, 2016-07-27:
Doesn't the checker run as soon as a new mission is loaded? Couldn't the _mission part in the navigator run the checker when it doesn't run using the on_inactive() call?

NaterGator, 2016-07-27:
@julianoes it does, but by that time the mavlink handler has already positively acknowledged the mission upload to the ground station. The proposed change allows the mavlink module to negatively acknowledge the mission upload if it fails feasibility checks which will allow for a more predictable UX on the ground station.

julianoes, 2016-07-27:
@NaterGator Hm good point! Ok, in that case I can see a clear benefit of converting the checker to a lib call.

closed 2018-02-21 by PX4BuildBot