PX4/PX4-Autopilot #5134: Check Pixracer ESC calibration

julianoes, 2016-07-24:
After a [report on the mailinglist](https://groups.google.com/forum/#!topic/px4users/V5lGnBgwY0Y) about, I tried to do an ESC calibration with Pixracer and found the following behaviour:

1. Calibration is started in QGC, battery plugged in is detected but the ESC doesn't beep for the calibration as it should.
2. After the calibration was done, the ESC starts beeping (probably because of no pulse) even though `pwm info` shows a disarmed pulse of 900us.

The results seemed to be the same with or without using the safety button before trying to calibrate.

I'll have to dig deeper with a scope soon.

julianoes, 2016-07-25:
I checked the Pixracer using a scope and I see that the pulse stays at 900us throughout the calibration process before it switches to no pulse.

I also checked on a Pixhawk where it works as expected.

julianoes, 2016-07-25:
This wasn't implemented, support added in #5137.

LorenzMeier, 2016-07-28:
Fixed.

closed 2016-07-28 by LorenzMeier