PX4/PX4-Autopilot #5135: SITL broken

AndreasAntener, 2016-07-24:
@jgoppert looks like your last commit broke SITL on OS X and CI.

On Mac I have the following issue building gazebo (catkin required?):
```
-- The C compiler identification is AppleClang 7.0.2.7000181
-- The CXX compiler identification is AppleClang 7.0.2.7000181
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.1") 
CMake Error at CMakeLists.txt:16 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!

```

On Jenkins it's different (seems the script directory has changed?):
```
-- Build files have been written to: /job/Firmware/build_gazebo
[  2%] [  5%] Scanning dependencies of target sdf
[  7%] Running C++ protocol buffer compiler on msgs/HilGps.proto
[ 10%] Running C++ protocol buffer compiler on msgs/MotorSpeed.proto
Running C++ protocol buffer compiler on msgs/HilSensor.proto
Generating /job/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
[ 13%] [ 15%] [ 18%] Running C++ protocol buffer compiler on msgs/SensorImu.proto
Running C++ protocol buffer compiler on msgs/HilControl.proto
Running C++ protocol buffer compiler on msgs/Float.proto
python: can't open file '/job/Firmware/Tools/sitl_gazebo/tools/xacro.py': [Errno 2] No such file or directory
make[3]: *** [/job/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf] Error 2
make[2]: *** [CMakeFiles/sdf.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[ 21%] [ 23%] [ 26%] Running C++ protocol buffer compiler on msgs/opticalFlow.proto
[ 28%] Running C++ protocol buffer compiler on msgs/lidar.proto
Running C++ protocol buffer compiler on msgs/CommandAttitudeThrust.proto
[ 31%] Running C++ protocol buffer compiler on msgs/CommandMotorSpeed.proto
[ 34%] Running C++ protocol buffer compiler on msgs/CommandRateThrust.proto
[ 39%] [ 36%] Running C++ protocol buffer compiler on msgs/CommandVelocityTrajectory.proto
Running C++ protocol buffer compiler on msgs/CommandRollPitchYawrateThrust.proto
Running C++ protocol buffer compiler on msgs/CommandTrajectory.proto
Scanning dependencies of target mav_msgs
[ 42%] [ 44%] Building CXX object CMakeFiles/mav_msgs.dir/HilGps.pb.cc.o
[ 47%] [ 50%] Building CXX object CMakeFiles/mav_msgs.dir/HilSensor.pb.cc.o
Building CXX object CMakeFiles/mav_msgs.dir/SensorImu.pb.cc.o
Building CXX object CMakeFiles/mav_msgs.dir/HilControl.pb.cc.o
[ 52%] [ 55%] Building CXX object CMakeFiles/mav_msgs.dir/Float.pb.cc.o
Building CXX object CMakeFiles/mav_msgs.dir/opticalFlow.pb.cc.o
[ 57%] Building CXX object CMakeFiles/mav_msgs.dir/lidar.pb.cc.o
[ 60%] Building CXX object CMakeFiles/mav_msgs.dir/CommandAttitudeThrust.pb.cc.o
[ 63%] Building CXX object CMakeFiles/mav_msgs.dir/CommandMotorSpeed.pb.cc.o
[ 65%] Building CXX object CMakeFiles/mav_msgs.dir/CommandRateThrust.pb.cc.o
[ 68%] Building CXX object CMakeFiles/mav_msgs.dir/CommandRollPitchYawrateThrust.pb.cc.o
[ 71%] Building CXX object CMakeFiles/mav_msgs.dir/CommandTrajectory.pb.cc.o
[ 73%] Building CXX object CMakeFiles/mav_msgs.dir/CommandVelocityTrajectory.pb.cc.o
[ 76%] Building CXX object CMakeFiles/mav_msgs.dir/MotorSpeed.pb.cc.o
Linking CXX shared library libmav_msgs.so
[ 76%] Built target mav_msgs
make[1]: *** [all] Error 2
make: *** [gazebo_build] Error 2
Build step 'Execute shell' marked build as failure
```

jgoppert, 2016-07-24:
@AndreasAntener sorry for that, should be fixed now, catkin is not necessary

LorenzMeier, 2016-07-24:
@jgoppert ```CMake Error at CMakeLists.txt:202 (install):
  install DIRECTORY given non-directory
  "/Users/lorenzmeier/src/Firmware/Tools/sitl_gazebo/models/.DS_Store" to
  install.```

LorenzMeier, 2016-07-24:
@jgoppert Part of the issue here are the submodules - one good approach is to merge the PR in the submodule first, then rebase the Firmware PR and wait for CI. That sequence allows to run CI through the submodule update.

Another option is to push the submodule change to a branch and point there. Then you can equally run the Firmware PR with the submodule working properly.

jgoppert, 2016-07-24:
@LorenzMeier did do that on a PR with a separate sitl_gazebo branch, the issue was CI was down, should have waited for it to go back up.

jgoppert, 2016-07-24:
Closed with #5136

closed 2016-07-24 by jgoppert

LorenzMeier, 2016-07-24:
@jgoppert I noticed some changes in the optical flow handling and a first pass looked like the flow integral was replaced with average flow - am I reading this correctly?

LorenzMeier, 2016-07-24:
Clarified it, I read things incorrectly.