PX4/PX4-Autopilot #5140: Manual control of ground based experimental setup using Mavros by bypassing pixhawk's low level controller.

RamArvinth, 2016-07-25:
Dear Developers, 

  We have an stationary experimental setup that consists of a beam rod whose centre is held by a base column. The joint is designed such that the rod has a 2 DOF. At each end of this rod is a motor, with only one end motor having vectoring capability by means of a servo controlled motor base mount. A platform at the centre of the rod holds the Pixhawk and other sensors. 

  The firmware had been cloned from the Px4/Firmware repository (https://github.com/PX4/Firmware.git). Based on the experimental setup configuration, appropriate mixers were created. The modified firmware works fine in terms of behaviour in Roll and Yaw with direct controls from the RC proving appropriate mixers are being used. Next, with this being stationary setup, by using 'mavros' we were able to achieve OFFBOARD control and perform trajectory tracking by publishing on the topic "/mavros/setpoint_attitude/attitude" and "/mavros/setpoint_attitude/att_throttle".

**Objectives**: 
	
  Right now, what we are doing with OFFBOARD mode is commanding varies attitudes (roll and yaw) and Pixhawk follows it using its onbaord low level controller. That means the controller for now is running on pixhawk. What we want to achieve is to be able to run controller on our PC so that at the end we can control our setup using different control algorithms. Our aim in the closed loop control is to directly get PWM (or RPM) values from our controller and forward these values to each motor via output pins of pixhawk. That is, we would like to find out a way to bypass pixhawks low level controller so that the ESCs instead of taking Pixhawk commands, take our command directly. 

**Problems**:

In order to achieve this, I have gone through all the topics of mavros package and think "/mavros/RC/OverRide" to be the most revelant but I'm still not sure of how to make use of it. Also i could not find any example for this particular topic so that I can use it for my purpose.

 Please, do suggest me if there is any other way this could be achieved if not by the above method.

 To Summarize - I want to bypass the Pixhawk low level controller and be able to command the motors from the controller on my system using mavros.

  I apologize if this issue is not relevant here, but I am not able find answers any where else and I could really use your help in achieving this.

Thanks!!

mhkabir, 2016-07-25:
https://pixhawk.ethz.ch/mavlink/#SET_ACTUATOR_CONTROL_TARGET

This is supported as a special case of offboard mode. You'll need to set the mixers to passthrough.

RamArvinth, 2016-07-26:
Hello @mhkabir , Thank you so much for your reply.
  
 As you mentioned, In order to make use of SET_ACTUATOR_CONTROL_TARGET, I define a mixer configuration which lets manual passthrough of the inputs. The mixer file I use looks like this,

Passthrough mixer for PX4IO
This file defines passthrough mixers suitable for testing.

Channel group 0, channels 0-7 are passed directly through to the outputs.

M: 1
O: 10000 10000 0 -10000 10000
S: 0 0 10000 10000 0 -10000 10000

M: 1
O: 10000 10000 0 -10000 10000
S: 0 1 10000 10000 0 -10000 10000

M: 1
O: 10000 10000 0 -10000 10000
S: 0 2 10000 10000 0 -10000 10000

M: 1
O: 10000 10000 0 -10000 10000
S: 0 3 10000 10000 0 -10000 10000

M: 1
O: 10000 10000 0 -10000 10000
S: 0 4 10000 10000 0 -10000 10000

M: 1
O: 10000 10000 0 -10000 10000
S: 0 5 10000 10000 0 -10000 10000

M: 1
O: 10000 10000 0 -10000 10000
S: 0 6 10000 10000 0 -10000 10000

M: 1
O: 10000 10000 0 -10000 10000
S: 0 7 10000 10000 0 -10000 10000

  Now, using this I could command desired Roll, Pitch, Yaw and Throttle values, directly. However, what I'm hoping to achieve is to have direct control over the PWM (or RPM) values commanded by my controller that I can directly pass over to the EScs (connected to pixhawk) through mavros, instead of commanding the attitude angles (Roll, Pitch, etc.).

- So far I have gone through various msg types available, but I haven't be been able find any examples on how these msg types can be implemented. Do let me know if any examples or any other documentation on how to implement this.

 - Since I do not have experience in python scripts, I have been working only on ROS. Using this, I create topics to publish my commands from the controller. But how do I make sure that the pixhawk subscribes to the commands being published?

   I'm looking to find specific mavlink msg type or mavros topics to which I could publish the commands (PWM or RPM values) in order to directly access the output pins of the pixhawk, which in a way would facilitate the bypassing of pixhawk's low level controller.

   I hope I was clear enough in explaining what exactly I am trying to achieve. I really appreciate all your help and guidance.

Thanks and Regards,
Ram

mzahana, 2016-11-05:
@RamArvinth Where you able to use rc override with PX4?

RamArvinth, 2016-11-05:
Hello @mzahana , 
   
   No, I wasn't able to achieve exactly as intended to. But as mhkabir mentioned above, I used the set_actuator_control to command the values of throttle, roll, pitch, yaw directly in OFFBOARD mode through mavros from my system. This anyway helped me avoid the use of a RC.

mzahana, 2016-11-05:
But does mavros rc/override works any way but you could not use it properly, or it does not work at all?

RamArvinth, 2016-11-05:
Technically it should work, but I couldn't get it work properly and haven't figured out why yet.

mhkabir, 2016-11-05:
No it's not supposed to. RC override is a horrible way to achieve what you
want to do.

fjctp, 2017-02-28:
Hi @RamArvinth,

I am trying to archive the same thing, controlling the motor directly. I wonder if you have any success?

I am using SET_ACTUATOR_CONTROL_TARGET ([link](https://pixhawk.ethz.ch/mavlink/#SET_ACTUATOR_CONTROL_TARGET))and group mixing 3 ([link](https://dev.px4.io/concept-mixing.html#control-groups)), but it does not work. I am still controlling roll/pitch/yaw/attitude only, not the motors.

After some digging, I think this maybe related to  #4029. I wonder if the fix is merged.

Thank you,
Jason

RamArvinth, 2017-03-02:
Hello @fjctp ,

   I was able to achieve something partly to what I wanted to actually do.

  Using help of contents from this [link](https://pixhawk.org/dev/ros/ground_rover), following a similar approach, i was able to set the throttle value _(still not the exact required rpm though)_ for motor.

  In addition, you can set an offset in the mixer file for the throttle channel such that when you set throttle value using rosparam, you have [0 to 1] as the range instead of [-1 to 1] if you probably would require only one direction of rotation.

  I'm not sure if this was of any help. But you could check if this suits your purpose.

Welcome!
Ram Arvinth

dagar, 2017-04-12:
I think we can close this? Please reopen if it's not resolved.

closed 2017-04-12 by dagar

mandelyoni, 2018-01-26:
Hi, how can I command the values of throttle, roll, pitch, and yaw directly to the PX4 using mavros?

Thank you

suraj2596, 2019-11-28:
Setting RPMs is equivalent to setting the PWMs of the ESCs which can be done using [this](https://mavlink.io/en/messages/common.html#SERVO_OUTPUT_RAW). Or am I missing something?