PX4/PX4-Autopilot #5141: MIN_PWM automatically resets to 1075 after reboot

garethdicker, 2016-07-25:
Trying to set MIN_PWM to 1000 instead of 1075, but after reboot MIN_PWM resets to 1075 with the airframe Generic Quad X selected.

Anyone know where the overriding of this param is happening and why?

julianoes, 2016-07-26:
How did you set it?

kd0aij, 2016-07-26:
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.mc_defaults#L38

garethdicker, 2016-07-29:
thanks, closing.

garethdicker, 2016-07-29:
Reopening because if I change 

https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.mc_defaults#L40

to be 1000 instead of 1075, it does decrease the min PWM, however then in QGC I can no longer change any parameters and receive the warning:

"parameter write failed: comp:1 param:"

The failure to set params in QGC only seems to happen by changing that particular line in rc.mc_defaults. Is it not correct to simply write:

if param compare PWM_MIN 1000
then
	param set PWM_MIN 1000 #no longer 1075
fi

?

closed 2016-10-23 by LorenzMeier