PX4/PX4-Autopilot #5143: GPS fusion timeout while replaying SITL logs

CarlOlsson, 2016-07-26:
There seem to be something broken in replay on current master 4fa59c3cde0fc0ad5d015c91b186d564894fb321

If I record a replay log in SITL with standard VTOL and replay the log file I get GPS fusion timeouts

log is here:
http://logs.uaventure.com/view/zTZEXPSzkPgdN7YdG7eA7R
```
pxh> INFO  [ekf2_replay] Replay in progress... 

INFO  [ekf2_replay] Log data will be written to ./build_posix_sitl_replay/src/firmware/posix/rootfs/replay_replayed.px4log

INFO  [lib__ecl] EKF using pressure height - commencing alignment
INFO  [lib__ecl] EKF alignment complete
INFO  [lib__ecl] EKF gps is good - setting origin
INFO  [lib__ecl] EKF commencing GPS aiding
WARN  [lib__ecl] EKF GPS fusion timout - resetting to GPS
WARN  [lib__ecl] EKF GPS fusion timout - resetting to GPS
INFO  [ekf2_replay] Done!
```

CarlOlsson, 2016-07-26:
[image]

CarlOlsson, 2016-08-02:
Here is a comparison of exactly the same mission flown in SITL, on with replay logging enabled and replayed and the other standard logging.

**Standard**
[image]
[image]

**Replayed**

CarlOlsson, 2016-08-02:
@tumbili 
This is the plot from a standard sitl flight with EKF2_MAG_TYPE = 1
[image]

CarlOlsson, 2016-08-02:
It looks like SITL is broken on master 033e9e140d0fb38d4cf85441682a65fcae0723f8
Here is the log file:
http://logs.uaventure.com/view/6SBJD4LkNhnmCLVK4ZiRHc

CarlOlsson, 2016-08-09:
It seems like there are several issues present here.

1) Position innovations diverge in a **standard logging** SITL mission

2) While replaying a RPL log generated in SITL the same thing happens but I also get warning messages  about GPS fusion timeouts.

3) GPS fusion timeout also occur while replaying logs generated from a physical platform.
Here is a plot from a flight with standard logging
[image]
Everything looks good except the known issue of switching gps module
Here is a plot of almost exactly the same flight done directly afterwords and then replayed.
[image]
I have noticed that if I set 
EKF2_GPS_V_NOISE = 1.5
EKF2_GPS_P_NOISE = 1.5
It looks a bit better for some reason:
[image]

@tumbili @LorenzMeier If you have any suggestions on how to debug this please let me know

CarlOlsson, 2016-08-11:
Summery of tests so far, the [same mission](https://github.com/PX4/Firmware/files/413579/mission.txt) was flown in all tests

Replayed logs generated in SITL on my computer, estimator diverges
[image]

Replayed logs generated in SITL on our beast (12 core computer) unclear what the problem is
[image]

Standard logging from 3 different laptops, all running Ubuntu 14.04 LTS
[image]
[image]
[image]

Standard logging from the beast
[image]

Please tell me if anything is unclear, if I should perform more tests or if you have any idea of where to start look for the issues.

tumbili, 2016-08-12:
@CarlOlsson I can look into this but it must clearly be related so SITL. I'm using replay regularly on master and no issues.

tumbili, 2016-08-12:
@CarlOlsson Clearly you can see in the plots that there are two resets taking place (position / height jump).
That is the reason why you are getting the gps timeout messages. The estimator refuses gps measurements and after the variance of the corresponding states has grown and reached the threshold the filter does a reset.

CarlOlsson, 2016-08-12:
Thanks @tumbili  for the info that you are using replay on a real system, that is very relevant since I can't. See point 3 above. Could you upload a replay.px4log that you can replay and I can try it on my computer?

tumbili, 2016-10-07:
@CarlOlsson Please post the log :-)

CarlOlsson, 2016-10-07:
Hehe ops, here it is:
http://logs.uaventure.com/view/K3fbyTrttBLojq3KjQ2Pk7

tumbili, 2016-10-09:
@CarlOlsson Hi Carl, when I try to replay your log on master then I get this

```
INFO  [lib__ecl] EKF using pressure height - commencing alignment
INFO  [lib__ecl] EKF alignment complete
INFO  [lib__ecl] EKF gps is good - setting origin
INFO  [lib__ecl] EKF commencing GPS aiding
ERROR [lib__ecl] EKF airspeed fusion numerical error - covariance reset
WARN  [lib__ecl] EKF GPS fusion timout - resetting to GPS
WARN  [lib__ecl] EKF GPS fusion timout - resetting to GPS
INFO  [lib__ecl] EKF baro hgt timeout - reset to baro
INFO  [lib__ecl] EKF baro hgt timeout - reset to baro
INFO  [lib__ecl] EKF baro hgt timeout - reset to baro
INFO  [lib__ecl] EKF baro hgt timeout - reset to baro
INFO  [lib__ecl] EKF baro hgt timeout - reset to baro
INFO  [lib__ecl] EKF baro hgt timeout - reset to baro
INFO  [lib__ecl] EKF baro hgt timeout - reset to baro
WARN  [lib__ecl] EKF GPS fusion timout - resetting to GPS
WARN  [lib__ecl] EKF GPS fusion timout - resetting to GPS
WARN  [lib__ecl] EKF GPS fusion timout - resetting to GPS
INFO  [lib__ecl] EKF baro hgt timeout - reset to baro
INFO  [ekf2_replay] Done!
```
Either something is wrong in the replay or the airspeed fusion does something strange. We can have a closer look at this together.

CarlOlsson, 2016-10-10:
@tumbili Hi Roman, If I replay the file with current master of both firmware and ecl I get:
```
INFO  [lib__ecl] EKF using pressure height - commencing alignment
INFO  [lib__ecl] EKF alignment complete
INFO  [lib__ecl] EKF gps is good - setting origin
INFO  [lib__ecl] EKF commencing GPS aiding
INFO  [ekf2_replay] Done!
```
If I do the same but with EKF2_ARSP_THR = 6.0 I get:
```
INFO  [lib__ecl] EKF using pressure height - commencing alignment
INFO  [lib__ecl] EKF alignment complete
INFO  [lib__ecl] EKF gps is good - setting origin
INFO  [lib__ecl] EKF commencing GPS aiding
WARN  [lib__ecl] EKF GPS fusion timout - resetting to GPS
INFO  [ekf2_replay] Done!
```

Since we have been flying with the airspeed fusion hundreds of times without problems I would say it is a replay issue, however will probably have time to check on the exact same system and the same code later today. If everything is fine when using normal logging we should definitely meet up and have a look :)

tumbili, 2016-10-10:
@CarlOlsson Sorry, I must have been using a wrong version of master, I noticed that the condition to fuse airspeed was always set to false. I need to try again.

priseborough, 2016-10-26:
@CarlOlsson, the log you provided had EKF2_ARSP_THR set to 0 which disables airspeed fusion. When I run the replay without airspeed fusion, the performance is marginal with large magnetometer innovations and position innovations, but the position does not diverge.

position innovations
[image]

magnetometer innovations
[image]

This is probably due to a combination of large magnetometer interference - see large variation in field length:
[image]

And large g transients which are exceeding the GPS motion filter manoeuvre design limit of 4g:
[image]

When I run it with an airspeed threshold of 10 m/s, the position does diverge. Inspecting the GPS speed and airspeed it looks like the airspeed sensor is very badly calibrated. 
[image]

A loss of velocity and position accuracy due to the large airspeed errors has been observed in-flight on this airframe before. See: https://github.com/PX4/ecl/issues/205.

Updating the ecl reference to latest master which includes a patch to prevent airspeed fusion affecting non-wind states unless dead-reckoning (see: https://github.com/PX4/ecl/pull/206 ) eliminates the loss of position fusion. These are the position innovations with EKF2_ARSP_THR = 10.0

[image]

These are the wind estimates:
[image]

priseborough, 2016-10-26:
I could not find any evidence that the replay was handling airspeed fusion incorrectly. The GPS rejection in the log was due to bad sensor data and a vulnerability in the ecl EKF which has been fixed.

closed 2016-10-26 by priseborough

CarlOlsson, 2016-11-13:
@priseborough 
Hi Paul,

I have now flown the Phantom wing maybe 15 times and the estimator is performing all fine. However, as soon as I do a replay log and replay it the estimator diverges, in the way I have explained in this issue. Could you please check if this only happens for me?
Replay log [here](http://logs.uaventure.com/view/SFtP25pmMJwTwRNwYZtoxd).

Normal logging
[image]

Replayed flight
[image]

priseborough, 2016-11-14:
What version of the software were you using for the flight? The EST5 message labels are incorrect.

CarlOlsson, 2016-11-14:
@priseborough 
Thanks Paul,

The log is the same as the one I posted [here](https://github.com/PX4/Firmware/pull/5763)
So it was with latest ecl master but also with the sidesilp in. That is strange because if I replay it with 
EKF2_GPS_CHECK = 0 the estimator still "diverges":
[image]

EKF2_GPS_CHECK = 21
[image]

Does it look differently for you?

priseborough, 2016-11-14:
I am testing using master at 5ad329b with EKF2_GPS_CHECK = 0 and it is not resetting the position. I always do a make clean before running replay to wipe the old replay_params.txt file. Please provide a commit hash if you want me to confirm results.