PX4/PX4-Autopilot #5149: Log dir naming unclear

sanderux, 2016-07-27:
@dagar i think you made a change to the dir naming of the logs.
It now looks like this:

```
drwx------ 2 sander sander 32768 jul 18 20:00 2016-0~1
drwx------ 2 sander sander 32768 jul 19 19:02 2016-0~2
drwx------ 2 sander sander 32768 jul 20 20:38 2016-0~3
drwx------ 2 sander sander 32768 jul 21 19:10 2016-0~4
drwx------ 2 sander sander 32768 jul 26 19:14 2016-0~6
drwx------ 2 sander sander 32768 jul 25 19:41 2016-07-25
drwx------ 2 sander sander 32768 jul 18 20:25 sess001
drwx------ 2 sander sander 32768 jul 18 20:32 sess002
drwx------ 2 sander sander 32768 jul 18 20:32 sess003
drwx------ 2 sander sander 32768 jul 19 19:03 sess004
drwx------ 2 sander sander 32768 jul 21 20:31 sess005
```

dagar, 2016-07-27:
Wasn't me.

This commit maybe?
https://github.com/PX4/Firmware/commit/c4cb916afaebe457ee86806fb4cafbd942469419#diff-5e74ae83e561d42b76884c376a51715a

@julianoes FYI

julianoes, 2016-07-29:
Is this on Pixhawk/Pixracer?

sanderux, 2016-07-29:
Hawk

julianoes, 2016-07-29:
Ok, I'll try to reproduce it.

julianoes, 2016-08-01:
I can't reproduce this in my recent flight tests.

@sanderux Is the card FAT32 partitioned? And can you try to repartition and try again?

sanderux, 2016-08-01:
i'll try

dagar, 2016-11-08:
Resolved?

sanderux, 2016-11-08:
the sess dirs are produced when no gps is available and thus no date is known.

closed 2016-11-08 by sanderux