PX4/PX4-Autopilot #5150: ms5611: bug in temperature correction code

KIrill-ka, 2016-07-28:
ms5611.cpp:
Following line does not work because dT is int32
 ```
    int32_t T2 = POW2(dT) >> 31;
```
Also there is some "loose" type conversion:
`	int64_t f = POW2((int64_t)_TEMP - 2000);`
 and
`	int64_t f2 = POW2(_TEMP + 1500);`

LorenzMeier, 2016-07-28:
What is your proposed fix and what result do you get out of the tested changes?

KIrill-ka, 2016-07-29:
Declare dT as int64_t.
and replace POW2(_TEMP + 1500) 
with POW2((int64_t)_TEMP + 1500).
Testing is not easy. The fact is that second order temperature compensation proposed in ms5611 datasheet is implemented incorrectly. int32 >> 31 is either 0 or -1.

LorenzMeier, 2016-07-29:
How do you know its incorrect? 0 or -1 can be a valid sign operation. I would like to understand how you determine that the code is actually wrong.

Same goes for POW2 of _TEMP + 1500. That will be a maximum of 1600 * 1600, which is far less than the max 32 bit integer.

So please make either a mathematically sound and complete argument (e.g. by a counter-example) or provide test data. I can't follow the analysis yet.

KIrill-ka, 2016-07-29:
> How do you know its incorrect? 0 or -1 can be a valid sign operation.
Datasheet doesn't state that the result should be trimmed to int32. It states T2 = dT^2 / 2^31. For me it's obvious that they didn't mean it should be 0 or -1.
> Same goes for POW2 of _TEMP + 1500. That will be a maximum of 1600 * 1600
What is 1600? The minimum temp for ms5611 looks like -40. _TEMP will be -4000+1500 = -2500.
It fits 32 bit, not a problem. You're right. Then  POW2((int64_t)_TEMP - 2000) still fits int32, int64 conversion is not needed.

closed 2016-12-26 by LorenzMeier