PX4/PX4-Autopilot #5153: HDOP/VDOP not being reported from GNSSv2

Aperture-Laboratories, 2016-07-28:
H/VDOP not being displayed in the GPS window of QGroundControl 2.9.7b/3.0.0. 

@mhkabir Mentioned that the CAN Driver may have not been updated when there were changes made to the handling of H/VDOP reporting.

mhkabir, 2016-07-28:
@pavel-kirienko Could you please look into this

pavel-kirienko, 2016-07-28:
UAVCAN driver does not report HDOP and VDOP values indeed. Possible solutions are as follows:

1. Obvious solution - receive messages `uavcan.equipment.gnss.Auxiliary`, which contain DOP information. This solution will increase the memory footprint by about 200 bytes.

2. Approximate `hdop = sqrt(position_covariance[0] + position_covariance[4])` and `vdop = sqrt(position_covariance[8])`

3. We already receive PDOP from the bus, so we could approximate HDOP = VDOP = PDOP. This will yield more pessimistic values.

commit d42832a5 (2016-08-07): Fixed #5153
commit d12be196 (2016-08-18): Fixed #5153
closed 2016-08-18 by LorenzMeier