PX4/PX4-Autopilot #5158: Snapdragon Startup Problems

pzoom, 2016-07-29:
Hello

I've got a Snapdragon and I followed the instructions at dev.px4.io to set it up, and I've put the latest 3.1.1 BSP and ADSP firmware on it, but I get the errors shown in the attached screenshot about "krait update param ### failed" and "ERROR: FastRpcWrapper Not Initialized"

I don't know how important / severe these problems are - I can connect to the board with QGC via Wifi, but can't calibrate the sensors (I'm aware that may be a separate issue due to needing an external GPS receiver - saw that in the google group messages).

I don't get any output from the mini-dm; it runs and says "DMSS is connected. Running mini-dm" but nothing more.

So, I'm wondering what I need to do to get a clean startup, thinking that could be important for further steps down the line.

Many thanks.

[image]

twistxj, 2016-07-29:
Hi, I am having the same issue.  I'm not sure how to resolve it.

julianoes, 2016-07-29:
Hi, thanks for raising the issue.
Have you updated the adsp image after flashing the latest 3.1.1 BSP?

I'm referring to this step:
http://dev.px4.io/advanced-snapdragon.html#push-the-latest-adsp-firmware-files

twistxj, 2016-07-29:
I went through that step and pushed the adsp files (after loading bsp 3.1.1).  Is there a way to verify this was done correctly?  I see the adsp files on the board.

julianoes, 2016-07-29:
@twistxj that should be correct then.

Let me try to reproduce your issue.

pzoom, 2016-07-29:
Yes, where it says "download the files" and refers to to 8074 Firmware zip - that file doesn't exist on Intrinsyc's website anymore. I did however get the add-on zip file and follow those steps.

julianoes, 2016-07-29:
I just (again) updated both 3.1.1 BSP, and adsp image again and I can't reproduce your issue.

Are you sure that you have not already added px4 to the autostart and you try to run it yet again? Does it happen every time, even if you try a coupe of times (with some seconds of pause)?

If you can't get mini-dm to work, you can try nano-dm: https://github.com/kevinmehall/nano-dm.

twistxj, 2016-07-29:
I'm not sure what it means to add px4 to the auto start.  Can you clarify what that means? Or can you tell me how to check that?

julianoes, 2016-07-29:
@twistxj this one: http://dev.px4.io/starting-building.html#auto-start-px4

twistxj, 2016-07-29:
Oh okay, I have not done that part of the guide yet.

pzoom, 2016-07-29:
Julian - I haven't checked that so I'll do that as soon as I get back. I'm traveling right now and will rejoin the thread on Monday. 

Thanks for the help

pzoom, 2016-08-02:
Update - I've got px4 to start cleanly now.

Here's how: I went through the Upgrade Snapdragon steps (http://dev.px4.io/advanced-snapdragon.html) again, including upgrading the Linux image and updating the ADSP firmware. Then, instead of just doing a clean and rebuild in my px4/Firmware directory, I deleted it all and re-cloned it and rebuilt it. I think this was the key because the Linux image and ADSP files from Intrinsyc were the same as before. So this step (http://dev.px4.io/starting-building.html#compiling-on-the-console) was done as if for the first time, and then these steps repeated (http://dev.px4.io/starting-building.html#qurt--snapdragon-based-boards) and now it builds cleanly (see attached image). 

Also, for the first time, I'm getting lots of output from the mini-dm.

Cheers too all, thanks for the help.

[image]

closed 2016-08-02 by pzoom

julianoes, 2016-08-02:
@pzoom thanks for the feedback, and glad it's working. However, I'm not sure what could have triggered the failure but anyway it's something to try in the future.

Could you add the advice to the [troubleshoot section in the devguide](https://github.com/PX4/Devguide/blob/master/advanced-snapdragon.md#troubleshooting)?

pzoom, 2016-08-02:
Will do

VictorSu2, 2016-08-03:
Hi,
I have also the same problem. And i tried upgrade the BSP first and then ADSP firmware and delete all px4 firmware and clone and build again to clean the start up.
But it still shows the message:

> root@linaro-developer:/home/linaro# ls
> FTM_AP  dataman  mainapp.config  px4
> root@linaro-developer:/home/linaro# ./px4 mainapp.config
> sh: 1: cannot create /sys/devices/system/cpu/cpu3/cpufreq/scaling_governor: Directory nonexistent
> ERROR [platforms__posix__px4_layer] krait update param 297 failed
> ERROR [platforms__posix__px4_layer] krait update param 646 failed
> 
> ______  __   __    ___
> | ___ \ \ \ / /   /   |
> | |_/ /  \ V /   / /| |
> |  __/   /   \  / /_| |
> | |     / /^\ \ \___  |
> \_|     \/   \/     |_/
> 
> px4 starting.
> 
> ERROR [muorb] Initialize Error calling the uorb fastrpc initalize method..
> ERROR [muorb] ERROR: FastRpcWrapper Not Initialized
> INFO  [sdlog2] [blackbox] /root/log/2016-08-03
> INFO  [sdlog2] [blackbox] recording: 07_43_57.px4log
> INFO  [dataman] Unkown restart, data manager file '/home/linaro/dataman' size is 103090 bytes
> INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
> pxh> INFO  [mavlink] using network interface wlan0, IP: 192.168.1.159
> INFO  [mavlink] with broadcast IP: 192.168.1.255
> INFO  [mavlink] partner IP: 192.168.1.155

That's quite strange.