PX4/PX4-Autopilot #5167: quaternion.proto File not found while running make posix_sitl_default gazebo

pchatrath, 2016-07-29:
Hello, I got fresh copy from PX4/Firmware and installed gazebo6. However make posix_sitl_default gazebo doesnt work. It cannot find quaternion.proto & vector3d.proto files. What am I doing wrong?

Log:
```
quaternion.proto: File not found.
vector3d.proto: File not found.
SensorImu.proto: Import "quaternion.proto" was not found or had errors.
SensorImu.proto: Import "vector3d.proto" was not found or had errors.
SensorImu.proto:8:12: "gazebo.msgs.Quaternion" is not defined.
SensorImu.proto:10:12: "gazebo.msgs.Vector3d" is not defined.
SensorImu.proto:12:12: "gazebo.msgs.Vector3d" is not defined.
make[8]: *** [SensorImu.pb.cc] Error 1
make[8]: *** Waiting for unfinished jobs....
[ 19%] Running C++ protocol buffer compiler on msgs/lidar.proto
[ 21%] Running C++ protocol buffer compiler on msgs/CommandAttitudeThrust.proto
make[7]: *** [CMakeFiles/mav_msgs.dir/all] Error 2
make[6]: *** [all] Error 2
make[5]: *** [gazebo_build] Error 2
make[4]: *** [src/firmware/posix/CMakeFiles/gazebo] Error 2
make[3]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/all] Error 2
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
make[1]: *** [gazebo] Error 2
make: *** [posix_sitl_default] Error 2
```

LorenzMeier, 2016-07-30:
Let me look into that. What operating system are you using?

pchatrath, 2016-07-30:
Thank you. I am using Ubuntu 14.04 LTS. I tried to debug the error. I found quaternion.proto & vector3d.proto file at /usr/include/gazebo-6.6/gazebo/msgs/proto however those file aren't being detected somehow. I removed gazebo and tried a fresh copy of PX4/Firmware. But it doesnt work. However it worked on another system that I have. (Ubuntu 14.04 LTS)

LorenzMeier, 2016-07-31:
I fixed this on master. Please report back if an update does not fix it for you.

closed 2016-07-31 by LorenzMeier