PX4/PX4-Autopilot #5170: MANUAL CONTROL lost message on param set (and save)

julianoes, 2016-07-30:
Every time QGC sends a param value to Pixracer, I get the message "MANUAL CONTROL lost". Presumably something is blocking the manual control timeout check.

dagar, 2016-07-30:
Have you determined that it's not on the QGC side?

LorenzMeier, 2016-07-30:
I expect that the param change triggers a save operation or similar which is stalling the check. This should be fine if it happens in the right thread, however, we should make sure the previous architecture still matches all intentions.

julianoes, 2016-08-01:
@dagar it's not the QGC side. It also happens if you just do a `param save` using nsh.

julianoes, 2016-08-01:
@LorenzMeier this is what happens during `param save`:
https://github.com/PX4/Firmware/blob/master/src/drivers/stm32/drv_hrt.c#L459

julianoes, 2016-08-01:
Basically what happens is that the PPM encode on Pixracer looses sync because everything is locked while we write to the FRAM because the BUS is shared with the baro.

My changes make the locked times shorter and therefore alleviate the problem, however, the global locking to write params is still wrong, and as the [comment in the code](https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/param/param.c#L840) there suggests, only the SPI bus should be locked.

I filed a separate issue for that #5188.

PR #5187 (2016-08-01): param: lock the bus as short as possible
Since the FRAM and the baro are on the same bus on the Pixracer, we
currently need to lock down everything (instead of just this SPI bus)
for the time when the params are written.
Therefore, we need to keep this locking as short as possible.

This change makes the locking even shorter by moving all param_get and
param_name and param_size calls out of the lock.

This resolves #5170 for me.

@LorenzMeier please review.
closed 2016-08-01 by LorenzMeier