PX4/PX4-Autopilot #5172: ALTCTL broken on Pixracer

julianoes, 2016-07-30:
Recent test flights with branch feature_hysteresis (36299c59b4b4775f9688a4b8b8315dcc3f5c543d) made my copter with Pixracer drop altitude rapidely. The behaviour in ALTCTL was fine on Pixhawk.

dagar, 2016-07-30:
Which estimator? How's your pixracer mounted? Anything covering the baro? Can you post the log?

julianoes, 2016-07-30:
Estimator: q/inav
Pixracer on soft foam in Hobbyking case (but it's not a HobbyKing Pixracer).

http://logs.uaventure.com/view/s8Ss23Prppmk3E9qJm9gaM

dagar, 2016-07-30:
There's a pretty big disagreement between GPS and GPOS alt. Do these logs accurately show what happened? Nearly 75 degree pitch briefly in ALTCTL?

julianoes, 2016-07-30:
The crazy pitch is probably when it came down, not sure.

Presumably the usual integration of accel measurment range saturation or aliasing. I was just able to reproduce this by just holding the quad, giving it around hover throttle and it though it went up 10m.

julianoes, 2016-07-31:
One thing to consider is that the recent sensors.cpp changes moved from lowpass filtered values to the integral.

In stable, it is:
https://github.com/PX4/Firmware/blob/stable/src/modules/sensors/sensors.cpp#L1007

And in master:
https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L1101

This means that the situation is identical for ekf2 which was using the integral before, but different for q/inav because it now uses the integral which doesn't have the 2 pole Butterworth filter applied.

I'm still not sure though what this means in terms of aliasing.

LorenzMeier, 2016-07-31:
That's pretty bad and should not have gone in. I missed that in the review at the time.

mhkabir, 2016-07-31:
@jgoppert Does this affect LPE as well?

julianoes, 2016-07-31:
I see the same behaviour on this frame with stable (1.3.4), so it looks like it's frame specific. I'll have to do more testing.

LorenzMeier, 2016-07-31:
We traced this down to excessive offsets resulting from a bad mount. Waiting for Julian to show new logs with fixed mounts.

julianoes, 2016-07-31:
I replaced the foam pads with HK orange foam. The MPU9250 is no longer showing a big bias at high throttle. The real underlying cause is still not clear to me but the issue "resolved".

closed 2016-07-31 by julianoes