PX4/PX4-Autopilot #5179: parameters PWM_AUX_MIN,MAX not implemented

kd0aij, 2016-07-31:
Shouldn't they be referenced here?
https://github.com/PX4/Firmware/blob/master/src/drivers/px4fmu/fmu.cpp#L332

julianoes, 2016-07-31:
Are they later overwritten by the pwm command?

LorenzMeier, 2016-07-31:
@kd0aij Use the pwm command to set them during startup. See the main PWM how to do those. Because FMU is not necessary the aux port that seems the more scalable approach. Alternatively we could re-architect the scripts to not use any bash values at all and just rely on params and detect wether the FMU driver runs as AUX or MAIN.

kd0aij, 2016-07-31:
If I understand correctly, the parameters PWM_AUX_MIN and PWM_AUX_MAX are not used anywhere, and one must instead create a custom airframe which sets bash variables with those same names in the startup script. (rc.interface already looks for those variables)

If that is correct, then it seems those parameters should be deleted.

kd0aij, 2016-08-07:
@LorenzMeier any comments?

kd0aij, 2016-08-28:
closed pending further discussion

closed 2016-08-28 by kd0aij