PX4/PX4-Autopilot #5180: Re-tune softer defaults for copters

LorenzMeier, 2016-07-31:
@julianoes This is for the defaults discussion. We should make sure to not de-tune the controllers as much in the inner loops but to tune the user response.

The gains became more aggressive since v1.2.0

```
-PARAM_DEFINE_FLOAT(MPC_ACC_HOR_MAX, 4.0f);
+PARAM_DEFINE_FLOAT(MPC_ACC_HOR_MAX, 10.0f);
-PARAM_DEFINE_FLOAT(MPC_XY_P, 1.0f);
+PARAM_DEFINE_FLOAT(MPC_XY_P, 1.25f);
-PARAM_DEFINE_FLOAT(MPC_XY_VEL_P, 0.1f);
+PARAM_DEFINE_FLOAT(MPC_XY_VEL_P, 0.09f);
```

julianoes, 2016-07-31:
I'm not too familiar with the params above but I would certainly reduce MPC_XY_VEL_MAX from 8 to 5.

https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_params.c#L312

LorenzMeier, 2016-07-31:
You're looking at the MAX, not the DEFAULT, which is 5: https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_params.c#L283-L297

julianoes, 2016-07-31:
Aha!

LorenzMeier, 2016-07-31:
Ongoing discussion with user feedback:
http://discuss.px4.io/t/v1-4-1-release-testing/759/10

LorenzMeier, 2016-08-01:
@AndreasAntener @tumbili Another important test flight area which would help VTOLs with better overall understanding / tuning guides / defaults.

LorenzMeier, 2016-08-02:
@kd0aij @pkocmoud Thoughts?

pkocmoud, 2016-08-02:
I will try this tonight and feedback.

LorenzMeier, 2016-08-06:
Done on master.

closed 2016-08-06 by LorenzMeier