PX4/PX4-Autopilot #5186: taisitter gazebo link failed or continue communication lost

jilinggui, 2016-08-01:
make posix_sitl_default tailsitter_gazebo ,the gazebo start up but when link QGC , the qgc continuelly dispaly COMMUNICATION LOST，and  the system terminal dispaly：
INFO  [platforms__posix__px4_layer] simulator is slow. Delay added: 11845 us
INFO  [platforms__posix__px4_layer] simulator is slow. Delay added: 13518 us
INFO  [platforms__posix__px4_layer] simulator is slow. Delay added: 16737 us

what happen？
my system is ubuntu 14.04, and the use gazebo6.

julianoes, 2016-08-01:
Does this happen every time or just once? What are the steps to reproduce it?

jilinggui, 2016-08-01:
every time  i connect to QGC via udp, it always connect failed ,or after a few seconds it communication lost.system terminal always dispaly INFO [platforms__posix__px4_layer] simulator is slow. Delay added: 11845 us ..

julianoes, 2016-08-01:
Do you do anything other than the defaults?

julianoes, 2016-08-01:
What version are you on? I don't have this make target.

jilinggui, 2016-08-01:
sorry , it is "make posix_sitl_default gazebo_tailsitter",i made a mistake

julianoes, 2016-08-01:
```
INFO [platforms__posix__px4_layer] simulator is slow. Delay added: 11845 us
INFO [platforms__posix__px4_layer] simulator is slow. Delay added: 13518 us
INFO [platforms__posix__px4_layer] simulator is slow. Delay added: 16737 us
```

This just means that your computer can't keep up. If you see it way too often, you'll have to make sure nothing else runs on the computer or find a faster one I'm afraid.

julianoes, 2016-08-01:
Another indicator if the simulation is fast enough is the "Real time factor" shown in Gazebo in the bottom right.

I see the performance issues with the tailsitter on my machine too, but I don't have any communication problems.

Is the communication only lost or also regained?

jilinggui, 2016-08-01:
it never regained unless i disconnect it. in addtion, when i execute "take off "conmmand ,then set mode "mission", the vehicle becomme crazy.

julianoes, 2016-08-01:
so it only looses the communication once then? Do you still see output in QGC or not?

jilinggui, 2016-08-01:
my ubuntu run Under the virtual machine

jilinggui, 2016-08-01:
yes .when i set the mode "mission",QGC dispaly "Error during mission communicaiton with vehicle:vehicle did not respond to mission item communiction:MISSION＿ITEM" .

julianoes, 2016-08-01:
> yes .when i set the mode "mission",QGC dispaly "Error during mission communicaiton with vehicle:vehicle did not respond to mission item communiction:MISSION＿ITEM" .

Aha, we'll any chance something crashes in that case?

So for the rest it works, I assume?

jilinggui, 2016-08-01:
but most of the time I haven't had time to execute "mission"command ,the connection has lost.

mhkabir, 2017-04-03:
Stale.

closed 2017-04-03 by mhkabir