PX4/PX4-Autopilot #5188: Block SPI bus while using FRAM instead of everything on Pixracer

julianoes, 2016-08-01:
Since the FRAM and the baro are on the SPI bus on Pixracer, there needs to be some locking to prevent the two from crashing on the bus.

Currently, this is done by blocking "everything" while doing a `param save`, see:
https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/param/param.c#L840

As the comment suggests, it would be better to only lock the bus.

bkueng, 2016-08-02:
What I noticed recently is the following:
[image]
After a param change via QGC, the scheduling of sensor sampling is messed up for ~1/2sec. I think it's during the param save and probably related to disabling the IRQ's (the plot is with 623b99327d0ac9d47af applied).

julianoes, 2016-08-02:
@bkueng I agree, it's still far from optimal. I got more "manual control lost" messages yesterday. Presumably because I had changed more params from the default.

LorenzMeier, 2017-04-10:
We have improved this a bit already (current testing suggests performance is sufficient), but the better architecture is still pending.

closed 2017-10-04 by LorenzMeier