PX4/PX4-Autopilot #5190: Most up to date math library

mkschreder, 2016-08-01:
I'm considering switching to using "matrix" library for all my math stuff (from mathlib) but to my surprise I can not find something as basic as inverse function in it. Is matrix library being actively developed? I need to find a math library with an interface similar to mathlib but with cross platform support.

julianoes, 2016-08-01:
What about the Matrix lib:
https://github.com/PX4/Matrix/blob/65679fbcbb4b31a1a77f3dc5268ca78ca95b3bf3/matrix/SquareMatrix.hpp#L111

mkschreder, 2016-08-01:
Nvm about the inverse, I found it in the squarematrix class (although I think calling it inversed() instead of inv or I is a better naming convention). Looks legit so far. Gonna try porting my mathlib stuff to it and see how it goes.

closed 2016-08-01 by julianoes

mkschreder, 2016-08-01:
Added inversed() to the Matrix class for convenience.. 

https://github.com/PX4/Matrix/pull/26/commits/20e6e763948cbe7aa2951af44e2402390abb1d13