PX4/PX4-Autopilot #5193: QuadChute action for mission drift

sanderux, 2016-08-01:
Some VTOL systems can suffer from actuation problems in FW flight.
This could be related to CG, aileron travel or simply a nonalignment of the control surfaces.
The result is a fly-off of the vehicle. This is a users remark on gitter that probably had such an issue (but got lucky) 

> I have some issues with the navigation in autonomous plane mode. After the transition to plane, it should go straight to the next waypoint but it turned left and went very long then came back to my visible range.  I had also tried autonomous flight in just quad copter with the same gps module its worked pretty well.  Does anyone know why it's not reliable in plane mode? Appreciate your help!

A standard geofence / rtl would not work in this case as this would attempt an RTL in FW mode, and it is disabled by default.

For standard VTOL we could consider deploying QuadChute when the VTOL drifts off course or is unable to navigate. Any thoughts? And is there already a detection for such a condition? and if not, what would you propose as detection method?

@AndreasAntener @tumbili @dagar

dagar, 2016-08-01:
So step 1 would be some kind of generic flyaway detection perhaps in navigator or something more specific in the fw controllers.

sanderux, 2016-08-01:
I think the easiest way is to see if the position setpoint error is increasing over a period of time (10 secs or so).

dagar, 2016-08-01:
That sounds reasonable. One thing that immediately comes to mind is figuring out how it would fit in with engine failure detection.

tubeme, 2016-08-14:
How about fly away in Quad mode? Not unseen scenario. It is good to think about fly aways in general. Good idea is to use the QuadChute for the purpose.

LorenzMeier, 2016-12-28:
Closing as stale.

closed 2016-12-28 by LorenzMeier