PX4/PX4-Autopilot #5197: frequent rc signal loss, snapdragon with pixhawk (using snapdragon_rc_pwm)

dhirajdhule, 2016-08-02:
Setup: Delta8 PPM receiver --> Pixhawk --> snapdragon. 
I am frequently getting manual control lost errors in qgroundcontrol, which is sending the system into failsafe. 
I found that when I power Pixhawk from servo rail and VCC on TELEM1 (supplied from snapdragon) at that time I get lot of manual control lost errors, but at least I can see rc channels topic updated in qgc with delay.
(I understand that VCC in TELEM1 is actually output pin)
So I powered pixhawk with only servo power rail ( disconnected VCC in TELEM1) as only gnd, tx, rx should be sufficient for UART communication. But surprisingly the rc channel topic update rate drops to 0 and also system doesn't respond to any signal. Same thing happens no matter if I supply from power connector on pixhawk or usb cable. 
I double checked all my connections. Both snapdragon and pixhawk are common grounded properly even when powering them separately.
I had some success when I was powering snapdragon from pixhawk (removed snapdragon's power connector). But it did work for only two flights, now I am getting manual control lost errors in this setup as well.
Any idea where I might be going wrong?

julianoes, 2016-08-02:
Have you tried powering the Pixhawk using a power module? I'm not saying this is a solution, I'm just wondering because that would be the recommended setup.

Do you have anything else connected to the Pixhawk? On the connectors or on the servo rail?

dhirajdhule, 2016-08-02:
I tried powering pixhawk with 5V supply from power module, but then rc_channels topic died.
I have only 4 servos and a ppm receiver on the pixhawk. Apart from that only safety switch, nothing else. 
I also replaced the serial cable between pixhawk and snapdragon to rule out any connection issue, then powered pixhawk using telem1 vcc (from snapdragon), but again the link is randomly turning on and off. I am getting link loss for 500 to 2000 ms errors.

It had worked yesterday, though, but after 4 flights it has again started giving this problem. 
Also when I was flying yesterday, the manual link seemed latent. I could feel the latency in the response from vehicle to stick commands.
 
Is the snapdragon_rc_pwm and its counterpart on snapdragon stable and tested? Did you encounter such errors before?

julianoes, 2016-08-02:
> I have only 4 servos and a ppm receiver on the pixhawk. Apart from that only safety switch, nothing else. 

Servos pull way more than the power module can supply. I recomment to power it using the power module and an additional BEC connected to the servo rail powering the servos. Or doing the tests without servos connected.

dhirajdhule, 2016-08-02:
Ohh, sorry, by servos I meant ESCs.

julianoes, 2016-08-03:
> Is the snapdragon_rc_pwm and its counterpart on snapdragon stable and tested? Did you encounter such errors before?

I have not tested this part in a while because I just haven't flashed it since it always just worked. I'll give it a try, hopefullly this week, and let you know.

julianoes, 2016-08-04:
> Is the snapdragon_rc_pwm and its counterpart on snapdragon stable and tested? Did you encounter such errors before?

I retested latest master for the Pixracer and it works fine for me. Can you verify that the connection to your Rx is ok and that the Rx is continually on? Or could you try with another receiver?

LorenzMeier, 2016-08-04:
Chinese Pixracer builds have a signal level issue because they have an optional resistor populated. I need to check if the files are bad or why that happens.

julianoes, 2016-08-04:
@LorenzMeier he is using a Pixhawk though.

dhirajdhule, 2016-08-04:
Ok, I will use another RX. I am using pixhawk with Delta 8 ppm receiver. I will post my results after testing that.
Just FYI, I had used master built around 2 weeks before, flashed travis master build from qgroundcontrol.

dhirajdhule, 2016-09-12:
I have tried another delta8 reciever, spektrum dsmx satellite reciever on pixhawk. But it gives the same problem. I also tested it using another hardware (FlytPOD) running px4 code. But again same rc link loss issue. 
I already have replaced the wires between pixhawk and snapdragon. I don't think there is any issue with connections since the link loss starts randomly irrespective of any motion. 
Btw, I also tried to put satellite receiver on snapdragon's port J15 dev/tty-1 and used rc_reciever driver in px4.config. But it does not appear to work. I tried to debug using list_topics and listener command in pxh shell of the mainapp. I understand that this is a separate issue. 

Any clues, what is going wrong?

VictorSu2, 2016-09-23:
Hi, following the instruction here.
https://github.com/ATLFlight/ATLFlightDocs/blob/master/PX4.md
If want to use Spketrum as rc, flash the atlflight set up firmware and if want to use ppm as rc, flash the px4 set up firmware.

dhirajdhule, 2016-09-23:
Thanks Victor,

I had followed px4 instructions.
One question, if I use ATLflight can I still use pixhawk as pwm out device to connect ESCs? Because I don't have qualcomm's ESC board. 
Also the board that we are using is given to us by qualcomm and intrinsyc is not ready to give us support and the files. 
Thanks anyway.

VictorSu2, 2016-09-23:
Maybe you need to ensure that the pwm_out_rc_in is started in the config files. The pwm_out_rc_in is the app that outputs ESC Uart data and receive the mavlink RC packet on the port J13.
The px4.config is the start up script(just like rcS script for Nuttx based px4) for ADSP and the mainapp.config is the one for Krait. 
The pwm_out_rc_in should be started in px4.config instead of mainapp.config.

Thanks
Victor

dhirajdhule, 2016-09-23:
yes, I have made sure that pwm_out_rc_in is running. I am also able to get rc signal on snapdragon from pixhawk. But the issue is that this signal is not continuous. I am getting manual control loss error which normally arises due to lost/missing packets.

julianoes, 2016-09-27:
> @dhirajdhule But the issue is that this signal is not continuous.

Is the signal not continuous which comes from Pixhawk, or is the signal coming from the receiver not continuous?

Do you get RC loss when you keep moving a stick, so that the values always change?

dhirajdhule, 2016-09-27:
@julianoes the signal from pixhawk to snapdragon is not working properly. The 'manual control lost' error appears continuously irrespective of stick movement. I have tested two recievers, Delta8 with Frsky Tx, and spektrym 7 channel DSMX receiver. Both receiver work well when used on other systems with pixhawk. My best guess is that the UART connection between pixhawk and snapdragon has severe packet drop. But I am not sure how I can test it.

julianoes, 2016-09-27:
Have you verified that the output on the Pixhawk is correct? E.g. by adding a printf when it sends the mavlink message? You could also check the rate at which it is sent.

dhirajdhule, 2016-09-27:
I haven't verified the messages sent from pixhawk yet. I will check the rate though.

julianoes, 2016-10-10:
@dhirajdhule have you ever gotten to the bottom of this?

dhirajdhule, 2016-10-21:
yes!
Because of some inexplicable reason, when the device is powered up using the 12V input to power module, the UART port starts to cause problems, which leads to RC data lost/regained errors being thrown multiple times in a second. This issue gets resolved if 5V input is used instead of 12V. So I used another DC to DC converter before the power module shipped with snapdragon kit.

Ans I also had compiled latest master with instructions on dev.px4.io. This solved the issue with rc data loss. But the link still has little bit lag, which I can clearly feel in manual mode. 
Thanks a lot for your help!

closed 2017-01-29 by LorenzMeier