PX4/PX4-Autopilot #5203: Quadcopter: Add TPA break point and "Drop" Failsafe

Jesus343, 2016-08-02:
- Tpa breakpint helps reduce oscillations at higer throttle with more agressive pids. Cannot get a good tune as of now for racing. 
-"Drop" failsafe : when signal from transmitter is lost make drone drop. Many racing quadcopters don't have gps or other software/hardware enabled.

jgoppert, 2016-08-02:
@LorenzMeier  @kd0aij @pkocmoud What do you guys think about these changes for the racer?

TPA breakpoint is described here: https://github.com/cleanflight/cleanflight/blob/master/docs/PID%20tuning.md#tpa-and-tpa-breakpoint

@Jesus343 is coming from clean flight and I just set him up with a pixracer so he has a lot of good input we should try to apply to the racing aspects of PX4.

LorenzMeier, 2016-08-02:
I'm all for improving the racer experience and happy to make whatever changes are required for it. The only thing I don't want to do are uninformed changes with no value like blindly changing controller update rates without a system latency analysis.

jgoppert, 2016-08-02:
@LorenzMeier agreed, need to make sure any changes are justified. This is really about addressing thrust non-linearity, you can see how ArduPilot addressed it here.

https://github.com/ArduPilot/ardupilot/issues/1261
http://ardupilot.org/copter/docs/motor-thrust-scaling.html

LorenzMeier, 2016-08-02:
It would seem we should be able to generally model the effect based on the user simply setting his trust to weight ratio.

LorenzMeier, 2016-08-02:
And I don't think this is trust nonlinearity. It's the transfer function from trust percentage changes to resulting torque.

kd0aij, 2016-08-02:
I think "Throttle PID Attenuation" is a good idea, based on my experience,
but I just haven't gotten around to creating a PR for it. Same with the
"kill on RC loss". I've implemented and tested that for sbus input, but not
for the general case. Any suggestions on how to make that one general? I
found it difficult to follow the failsafe implementation, and my questions
about it are still unanswered.

jgoppert, 2016-08-02:
I think copying what ardupilot has done here, seems reasonable. Here are their latest parameters for various vehicles: https://github.com/ArduPilot/ardupilot/search?utf8=%E2%9C%93&q=MOT_THST_EXPO

Whatever the effect is caused by, modelling it as somewhere between a 1st and 2nd order phenomenon seems to work. @LorenzMeier can you explain your idea on the thrust to weight modelling approach?

LorenzMeier, 2016-08-02:
@kd0aij for motors off you just need to look for the auto disarm function in commander and add a condition for RC gone.

jgoppert, 2016-08-02:
@kd0aij  @LorenzMeier  on loss of manual control it would be nice to be able to configure it in the safety menu to either hover, land, or cut power.

LorenzMeier, 2016-08-02:
@jgoppert never mind, I meant a transfer function per motor plus a throttle to torque mapping. Starting with the transfer function would be a good start.

jgoppert, 2016-08-02:
If you use some simple propeller modelling you can get an idea of the relationship between rpm and thrust, just ignoring the motor transfer function for a moment and looking at the DC (steady-state) response.
http://adamone.rchomepage.com/calc_thrust.htm

[image]

[image]

kd0aij, 2016-08-15:
This simple change works very well on my Pixracer/S250 racer: https://github.com/PX4/Firmware/pull/5324

LorenzMeier, 2016-08-18:
@kd0aij There can be certain sensitivities involved with projects feeling that you take their ideas and/or violate their copyright (in particular if they strongly feel about copyleft licensing). I've found that although we're actually always trying to attribute (in the best academic tradition), that attribution is very often not appreciated and instead people claim ownership of ideas that are not even their own. So I would recommend to only attribute original work (meaning which is a first-time implementation of something with sufficient total innovation) and otherwise just sticking to text-book terms.

This here is a particularly good example where we really just applied PID gain scheduling (https://en.wikipedia.org/wiki/Gain_scheduling), a textbook technique not invented by Cleanflight and as old as PID controllers themselves. And of course any controls engineer will know that, but people lacking the background or community users won't. They will just see the three-letter acronym and (with their limited understanding) conclude it was taken from somewhere else.

From my experience we should strive to go back to the basic techniques and terms whenever possible (also for the sake of consistency with literature).

lthall, 2016-08-18:
For the record I will always support using any code or approach taken in Arducopter in any other open source code. Well, at least the code and ideas that I have heavily contributed to, so pretty much any of the Arducopter control chain code.

I would appreciate it if you would at least reference the approach "This is the same approach taken by the arducopter code" when introducing the new feature to the community, however I only consider this polite, not mandatory.

I would also appreciate it if you take the time to feed back anything you learn while using approaches I pioneered so we can all learn. Especially if you find a mistake!!!!

If anybody gets their nose out of joint then feel free to link them to this post or direct them to me personally.

(I would also recommend having a close look at our mixing and prioritising motor mixing code)

LorenzMeier, 2016-08-18:
Thanks Leonard! That's very sensible and much appreciated. It is how I would always operate in an academic context. My note stems from an offline discussion earlier today where I got contacted via Skype hinting in the opposite direction where referencing other people's work was already seen as a first step towards not going back to the basic technique and violating copyright and I was asked to raise the issue.
I found that in particular remarkable as James really just referenced work and concepts - him using the term "copy" in reference to the polynomial coefficients was already considered on the edge of copyright violation.

You can see that we're operating currently the way you're suggesting by referencing the project / person suggesting the use of gain scheduling for attitude control (in this case Cleanflight):
http://discuss.px4.io/t/pixracer-feedback-thread/27/158

We will review this policy if it turns out to be abused. That does of course never impact the attribution of true innovations, which is always due.

MAVProxyUser, 2016-08-19:
You taking any kick backs from Chris Anderson Lorenz? I heard he's been trying to kill Ardu* in favor of your project?

kd0aij, 2016-08-19:
@LorenzMeier @lthall For the record, my implementation of "PID gain scheduling" used parameter definitions similar to Cleanflight to avoid confusing any potential users, and to make it easier to compare tunings between Pixracer and Cleanflight setups. It never occurred to me that the approach might be original to either Cleanflight or Ardupilot.

MAVProxyUser, 2016-08-19:
Given that you deleted the comment without answering. I am going to take that as a "YES, I do take kickbacks from Chris Anderson". Thanks Lorenz.

LorenzMeier, 2016-08-19:
PX4 gets zero funding from 3DR, neither directly not indirectly. And we will continue to ensure discussions on our issue tracker meet professional standards.

kd0aij, 2016-08-19:
Test results posted here: http://discuss.px4.io/t/fpv-racer-tuning/969?u=mark_whitehorn
Please comment on that thread if you do any testing and refinement of the tuning parameters.

closed 2016-12-26 by LorenzMeier