PX4/PX4-Autopilot #5205: new logger: filename truncated in message file

kd0aij, 2016-08-02:
```
2016_08_02_14_12_03: ARMED by RC
2016_08_02_14_12_03: [logger] file: /fs/microsd/log/2016-08-02/14_12_0
2016_08_02_14_12_06: Takeoff detected
2016_08_02_14_12_14: LOW BATTERY, RETURN TO LAND ADVISED
2016_08_02_14_12_16: CRITICAL BATTERY, LANDING ADVISED!
2016_08_02_14_13_29: Landing detected
2016_08_02_14_13_29: DISARMED by RC
2016_08_02_14_13_34: ARMED by RC
2016_08_02_14_13_35: [logger] file: /fs/microsd/log/2016-08-02/14_13_3
```
looks like a 50 character limit somewhere

bkueng, 2016-08-02:
yes: https://github.com/PX4/Firmware/blob/master/msg/mavlink_log.msg
This will be changed, as we plan to add all PX4_WARN/ERR output to the log.

kd0aij, 2016-08-02:
OK, closing

closed 2016-08-02 by kd0aij