PX4/PX4-Autopilot #5210: EKF2 3D mag fusion diverges in SITL

tumbili, 2016-08-03:
Noticed this problem when flying the Solo in SITL. For some reason the mag field states (earth field and bias) are not correctly learned in the drone will eventually loose heading and start toilet bowling.
Forcing the filter to use the simpler heading fusion (EKF2_MAG_TYPE=1)  eliminates the problem.
@CarlOlsson Maybe this could be related to your replay problems. Can you confirm that 3D mag fusion is working on real hardware?

svpcom, 2016-08-03:
@tumbili I've have a flights using EKF2 (EKF2_MAG_TYPE = 0) and have no issues
http://discuss.px4.io/t/v1-4-1-release-testing/759/5?u=svpcom

Some time ago there was a bug with double sending mag data which broke EKF2 (it was fixed for HW to use only one magnetometer). May be this was not fixed for SITL

CarlOlsson, 2016-08-05:
@tumbili Yes mag fusion seems to work reliably on our hardware with EKF2_MAG_TYPE = 0

tumbili, 2016-08-05:
@CarlOlsson BTW, this is a replay logfile taken with an IRIS. 3D mag fusion seems to work well, so this is really a SITL bug.
http://logs.uaventure.com/view/uecoZwKugvVjqkrRdQyEGB

LorenzMeier, 2017-04-12:
Fixed

closed 2017-04-12 by LorenzMeier