PX4/PX4-Autopilot #5211: V1.4.1RC4: Landing detector not working properly in all situations

ecmnet, 2016-08-03:
After triggering auto-landing the vehicle landed on a slope surface. Doing so, I face a strange sequence of state changes:

[image]

julianoes, 2016-08-03:
Can you share a log for this please?

ecmnet, 2016-08-03:
Another short landing test shows that failsafe was triggered again, after landing was detected:

[image]

For this the log is: http://logs.uaventure.com/view/FLCCUajbQcMbRZYCqtJ8M8

julianoes, 2016-08-04:
I looked at all the data that could trigger a take-off detected and I can't see which one would have triggered it. If you can debug this further with some printfs that would be helpful.

ecmnet, 2016-08-05:
Maybe this issue occurs, because the vehicle is still wobbling (roll rate, pitch rate) on sloppy surfaces due to the relative high thrust during auto-landing.  The takeoff is detected about 1/2 second after landing. Maybe a time hysteresis would help here.

ecmnet, 2016-08-05:
@julianoes Additionally I've found out, that after auto-landing, the state is switched back to position hold (the state was in when triggering auto-landing):

[image]

Additionally RollRate and PitchRate were quite high during landing.

ecmnet, 2017-04-11:
Fixed for me with current master

closed 2017-04-11 by mhkabir