PX4/PX4-Autopilot #5212: V1.4.1RC4: LPE Position hold not stable

ecmnet, 2016-08-03:
@jgoppert Another issue regarding LPE: I left the vehicle unarmed for a while. After takeoff, I switched to althold, then to position hold. Both modes were quite unstable. Very interesting was position hold. The behavior was as if the pos gains were too high (massive control reaction in both directions, roll and pitch). (But when switching to pos hold shortly after a reboot, it works fine). Initially I noticed, that position as well as LPOS.Z have drifted a lot while being unarmed on ground (Bottom clearance was about 5.8m)

I assume, that raw flow information is compensated by altitude above ground. How is the altitude above ground determined? Is it dependent on terrain estimation? 

Log: http://logs.uaventure.com/view/iMqiBYS28xwjQH99CZjqBJ

Note also, that this did not occur with 1.4.1RC1.

jgoppert, 2016-08-03:
So this is without gps using flow outdoors?

jgoppert, 2016-08-03:
Your gains don't match the recommendations I made here to prevent excessive tilting when using optical flow position hold outdoors: http://dev.px4.io/optical-flow-outdoors.html

jgoppert, 2016-08-03:
@ecmnet If altitude above ground had drifted off and wasn't fixed on reset it could contribute to the problems you described. If it think it is higher above ground than it is, then position movements will be exaggerated in the estimator. The flow is basically: ground_distance_moved = altitude * flow_in_radians ( l = r*theta)

ecmnet, 2016-08-03:
@jgoppert This issue does not occur any more, when the reset of LPOS.Z is omitted in the way I mentioned in https://github.com/PX4/Firmware/issues/4878. Regarding the gains: Defaults work quite perfect.

I am closing this issue with reference to  https://github.com/PX4/Firmware/issues/4878.

closed 2016-08-03 by ecmnet