PX4/PX4-Autopilot #5213: NSH over Mavlink commands don't work

mhkabir, 2016-08-03:
@bkeung @julianoes 

```

mhkabir, 2016-08-03:
Also it doesn't seem to work over a UART (Telem2 @921600) : 
```python mavlink_shell.py /dev/ttyTHS0 -b 921600
Connecting to MAVLINK...
```
and NSH never turn up.

ecmnet, 2016-08-03:
@mhkabir I face no issues over UART: PX4 -> (UART 921600) -> MAVComm -> (UDP) -> MAVGAnalysis. Refer to screenshot [MAVGanalysis](https://camo.githubusercontent.com/f8fd213404e7a76485f4e9f958a7c5ddea527be2/68747470733a2f2f7261772e6769746875622e636f6d2f65636d6e65742f4d415647434c2f6d61737465722f4d415647434c2f73637265656e73686f74362e706e67)

mhkabir, 2016-08-03:
Hmm. Weird. I'll have to check the stack, likely. Because I'm also running UAVCAN.

bkueng, 2016-08-03:
Yes, the symptoms indicate that you ran out of RAM.

@ecmnet nice, you already added it to MAVGanalysis :+1:

mhkabir, 2016-08-03:
Nope, RAM's fine. Also, I got it working over the UART@921600. Bad cabling issue :P 
```
             total       used       free     shared    buffers     cached
Mem:       3946284     773076    3173208       2548      28372     362400
-/+ buffers/cache:     382304    3563980
Swap:            0          0          0
```

mhkabir, 2016-08-03:
OK can't repro over USB anymore, weirdly enough :P Will reopen if it crops up.

closed 2016-08-03 by mhkabir

huangwen0907, 2017-03-03:
@mhkabir Excuse me. Did you solve this bug? Now I face a same bug as yours. Hope you can help me. many many thanks~

davids5, 2017-03-03:
It would be that you have run out of RAM on the Flight controller not the host. You would need to run free on the pixhawk. Connect to the hardwired [debug console.](https://dev.px4.io/advanced-system-console.html) But I suspect it will not work if the system is that low on RAM.

tumbili, 2017-03-03:
I just tried mavlink shell on latest master on Pixhawk pro (the one that has more RAM) and it works fine.

TingKaiChen, 2017-12-05:
Hi everyone,
I find a same problem on my Pixhawk-RaspberryPi connection.
The connection is via USB (/dev/ttyACM0) with baudrate=115200.
The firmware of my Pixhawk is APM:Copter V3.5.4 (284349c3)
I use the MavProxy to connect to the Nuttx shell by this [link](http://ardupilot.org/dev/docs/interfacing-with-pixhawk-using-the-nsh.html), and enter the nsh successfully.
But when I use the command "mavlink start -d /dev/ttyACM0 -b 115200",
it shows "mavlink: command not found".
This error can be reproduce when I connect the nsh with QGroundControl.

The free memory is 63360, by using "free" command in nsh.
Other commands (ps, top, ls and free) can work perfectly, so I don't know if this is an "out-of-memory" problem.