PX4/PX4-Autopilot #5218: Sudden, inexplicable mode switch from AUTO to STAB

mswingtra, 2016-08-03:
As discussed at today's meeting, I'm creating an issue and sharing the log.

During one of our auto survey flights two weeks ago, the flight mode suddenly has switch from AUTO to STAB without any reason.
There was no mode change commanded either from RC or QGC.
Also, we did not observe any signal loss (RC, GPS, telemetry), and it occurred only once (we flew with the same codebase and hardware three more times at the same location).

Log is [here on LogMuncher](http://logs.uaventure.com/view/mr8Muek656VLwo2wKzUCtF).

We flew with master, commit fb1fd20.
QGC version used during that flight was release v2.9.7b.

dagar, 2016-08-03:
Well that's terrifying. It was a main state transition with no corresponding RC change.

I don't think we have any way to check if it was initiated by a QGC mode change. Do you think accidentally selecting the mode through the QGC drop down is a remote possibility? Even inadvertently opening and then clicking somewhere else intending to close it.

mswingtra, 2016-08-03:
To change it through QGC you have to click on the current mode, and then select a new one from the list, so it is hard to do accidentally and without noticing it.
At that time, I was monitoring the flight in QGC, and I do not recall doing it as I was just observing flight behaviour (without touching mouse or touchpad) since it was at the edge of line of sight. 
Of course, I am not excluding this possibility, but I simply to do not recall myself doing anything close to mode changing.

dagar, 2017-04-11:
Did you ever see this again? Unfortunately, I don't know that we can do much without more debugging information.

closed 2017-04-11 by dagar