PX4/PX4-Autopilot #5221: ROS / catkin build should build SITL app

LorenzMeier, 2016-08-03:
@jgoppert Is this quick to fix with your CMake-fu? We need to build SITL when Catkin builds.

jgoppert, 2016-08-03:
@LorenzMeier already on it here: https://github.com/PX4/Firmware/pull/5181

It is very close now, just creating the temporary file systems on the fly etc and have to figure out how best to handle paths.

LorenzMeier, 2016-08-03:
One key thing that we should ensure is that the install command does not become part of a normal build (which should go into devel), but copies / deploys what is present in devel to the host machine (or packages it into a distributable format doing the same).

This might be already the case, I have not entirely followed the diff. Its just important that we do not mix the machine-deployed version for end-users with the Firmware-local current WIP a developer is working on.

LorenzMeier, 2016-08-03:
I fixed it tentatively on master in 5b06b40ed9b1aa6823af0fabb774c0d3a514f98f. Includes your change.

closed 2016-08-03 by LorenzMeier