PX4/PX4-Autopilot #5226: Snapdragon UART to PWM ECS

twistxj, 2016-08-04:
Hi,
I loaded the latest firmware and adsp to my snapdragon board.  How do I enable ESC PWM output?

Thanks

julianoes, 2016-08-04:
For PWM output you currently need to wire it up to a Pixhawk, Pixracer, Pixfalcon, see the instructions here:
http://dev.px4.io/hardware-snapdragon.html#uart-to-pixracer--pixfalcon-wiring

You then need to set the param `SYS_AUTOSTART` of the Pix* to 20000.

so that it triggers passthrough mode:
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/20000_snapdragon_rc_pwm
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L836

I need to document this in the Devguide, let's keep this issue as a reminder for it.

Soon, however, PWM support will work with Snapdragon directly, see:
https://github.com/ATLFlight/dspal/pull/3

twistxj, 2016-08-04:
Thank you for that information.  

When you say "PWM support will work with Snapdragon directly" does that mean that no additional wire setup to a pixhawk, pixracer, pixfalcon will be required? 
 Also is there an estimated completion date for that dspal update?

julianoes, 2016-08-04:
> When you say "PWM support will work with Snapdragon directly" does that mean that no additional wire setup to a pixhawk, pixracer, pixfalcon will be required? 

Correct.

> Also is there an estimated completion date for that dspal update?

Not sure, ask in ATLFlight/dspal#3.

closed 2016-08-05 by LorenzMeier