PX4/PX4-Autopilot #5229: Commander: mission_finished true at mission start

sanderux, 2016-08-04:
On the FC (pixhawk),  mission_finished condition appears to be true before mission is started.
Under specific conditions (only data link and dll failsafe disabled) switching to mission mode is therefor not possible due to this line:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/state_machine_helper.cpp#L751

The behavior is not reproducible in SITL

@julianoes any idea why this bool is different outside sitl?

sanderux, 2016-08-04:
This would probably bypass the problem, but not solve it: https://github.com/PX4/Firmware/pull/5230

julianoes, 2016-08-04:
Where exactly is the condition true?

sanderux, 2016-08-04:
In this line, when switching to mission mode: https://github.com/PX4/Firmware/blob/master/src/modules/commander/state_machine_helper.cpp#L751

I conclude that condition to be true because;
- A valid mission existed
- Datalink was active but the datalink failsafe was disabled
- No rc was attached
- Failsafe switch to RTL when attempting to switch to mission
- Removing that line fixed the problem

julianoes, 2016-08-04:
What do you mean with:

> Failsafe switch to RTL when attempting to switch to mission

sanderux, 2016-08-04:
When (from QGC) i select flightmode "mission" it responds: "Failsafe on" and sets the mode to Return
This happens because of the code i qouted

julianoes, 2016-08-04:
I would argue that makes sense because you have no failsafe whatsoever. If you fly a mission and you are not using any datalink failsafe behaviour, the failsafe behaviour is undefined. E.g. when the mission is finished (loitering in the air) and the datalink is dead, what is it supposed to do?

sanderux, 2016-08-04:
I addressed this is my pr: https://github.com/PX4/Firmware/pull/5230
There are 2 scenarios;
1. link lost, mission finished and vehicle landed
2. link lost, mission finished and vehicle loitering in air

Scenario 1 is fine, the user had set no failsafe for links lost and obviously wants this behavior
Scenario 2 should be addressed, i therefor added the !landed condition to decide what to do.

With my PR it will:
Be OK with scenario 1
RTL in scenario 2

julianoes, 2016-08-04:
Yer I think I agree. Let me check the code.

closed 2016-08-05 by sanderux