PX4/PX4-Autopilot #5231: LPE: Estimated position should never reset

LorenzMeier, 2016-08-04:
@jgoppert After looking in more detail at the LPE behaviour I think the position reset is not what we want. Coordinates should never jump, not even in a filter reset. We need to address this before the release.

There is actually only a single consistent way that will not confuse users or lead to system crashes due to incorrect coordinates resulting from local coordinates being set before arming:

  * 1) Keep resetting the local position estimate to zero until a valid GPS or flow fix is present
  * 2) Once the GPS fix is present, start the normal estimation, which will lead to home being initialised
  * 3) If a filter reset occurs reset the local coordinates to the last known good position. Never reset local estimates in any shape in any situation

Please trust our operations knowledge with real users in the field on this one and lets get this release out!

LorenzMeier, 2016-08-04:
@ecmnet @art-mx FYI

tumbili, 2016-08-04:
@LorenzMeier I'm interested in the overall reset method (in terms of user experience) as well for ekf2.
I have this Pr up https://github.com/PX4/Firmware/pull/4741 which intends to handle a filter reset for different situations.
My and @priseborough thinking was to publish the corresponding delta values for the states which were reset so that the controllers can adjust their setpoints. However, whether the controller should apply the offsets to the setpoints is also dependent on the current flight mode.
E.g. in auto missions you would not want to shift the actual setpoints, because the plane should really be flying at the specified locating / altitude. However,  for most of the manual modes the setpoints should be shifted as the user would be confused if all of a sudden the drone starts moving to a different location / altitude.

LorenzMeier, 2016-08-04:
It makes sense to indicate that to the controller, however, that's a separate consideration from maintaining as accurate coordinates as possible. But you're right that we should address both eventually.

jgoppert, 2016-08-04:
I think if we handle the fault diagnosis correctly we can avoid the majority of resets. Besides a NaN this is the only time LPE resets and I agree we should replace it with some better logic. We know that if we haven't armed and the sonar data is bad then most likely we are still on the ground. This logic could be used to either lock out z estimation. Right now the vehicle altitude is free to estimate if baro initializes, and the terrain is free to estimate if the distance sensor initializes. Initialization here means we have received a series of good measurements with low variance. So the vehicle altitude is freely estimating and occasionally these sonar sensors give odd values on the ground, I've seen some report 800 cm 5 cm off the ground and that is very confusing to the estimator. If I get consistently 800 cm when not armed, we really should be able to ignore that too. Of course, there are always edge cases.

I think getting a series of test cases in SITL is the best approach here. It is very hard to think of all the edge cases without testing. You make a small change for one user and break it for another.

ecmnet, 2016-08-04:
@jgoppert Maybe my latest test are related: Test flight with latest master ( 00d4eae). Compared to 1.4.1RC4, I saw again jumps when switching to altHold: 

http://logs.uaventure.com/view/5LS3krHVSU5i2ZgHrZ2r7a#Local_Z_PLOT

LorenzMeier, 2016-08-04:
@jgoppert I much rather would stop supporting sonars and out of focus flow setups. You can't fix everything in software and if people get the setup wrong they have to fix it.

The reason stuff here is so sensitive is because you are trying to cater to those fundamentally broken setups. We should instead focus on plain GPS, then with flow in focus, then with lidar.

It will be hard to make LPE the general default if that's not the focus. We don't have the development bandwidth required to solve all sonar edge cases. Sonar will need to be figured out by people wanting to tinker with it, but separately from lidar.

I would of course also support if LPE want to focus on trying to be as good as possible with flow and indoors for developers and with support for ultrasound - but then it will be hard to make it the default for the system with the current development bandwidth we have on it.

jgoppert, 2016-08-04:
We really need to be able to see the dist_bottom states in addition to local Z @AndreasAntener have you been able to add that yet to logmuncher,  would be really helpful to quickly diagnose some of these user report.

jgoppert, 2016-08-04:
@LorenzMeier my main priority is GPS  outdoor flight as well right now, I have an outdoor flight coming up soon. So I don't plan to degrade performance.

I do hope that I can make it work for both cases. @ecmnet is so frustrated because the altitude hold was working very well for him before I started publishing the estimated vehicle height above sea level as local z instead of agl. So something is off there and trying to track it down.

art-mx, 2016-08-04:
Why does the estimator diverge in the first place? In my case I'm seeing it with an error free external estimation source (tachymeter), so the problem should be in the way estimator works?

LorenzMeier, 2016-08-04:
It does not diverge, it resets its estimate. We need to change that before the release

LorenzMeier, 2016-08-05:
Implemented in #5246

closed 2016-08-05 by LorenzMeier