PX4/PX4-Autopilot #5232: Issue connecting gazebo in SITL

krijnen, 2016-08-04:
Hi all,
I'm trying to install a gazebo simulation on a new computer. I've got it running on a laptop without problems, other than it's quite heavy. I tried to copy the setup to the new pc:
ubuntu 14.04
ros jade
gazebo 6.6
I'm using the manual start of the firmware, "no_sim=1 make posix_sitl_default gazebo", this shows the correct output. When i then launch gazebo with the copter, the px4 terminal starts outputting: 
"INFO  [platforms__posix__px4_layer] simulator is slow. Delay added: 193687 us"
multiple times a second.
On the laptop I'm running a slightly old version of the firmware, I tried both that version as well as the newest version of the firmware (editing my .bashrc file accordingly), both resulting in the same behaviour. 
Whereas on the laptop the gazebo simulation is slowed down to around 0.8, it shows 1.0 on the pc, so computer power doesn't seem to be the problem. I've seen a couple of similar problems, but none of the suggested fixes apply. Any ideas I could try?

LorenzMeier, 2016-08-07:
I'm assuming ```make posix gazebo``` works for you?

krijnen, 2016-08-08:
Hi Lorenz,
Running the standard simulation does indeed work. Its when running the modified sdf model that it doesn't. Undoubtly it has something to do with the setup of either plugins or dependencies on the desktop. Gazebo does load the correct model, and gives the same terminal output as on the working laptop (showing the sitl plugins are loaded and supposedly publishing). 
Thanks for your help.

krijnen, 2016-08-22:
For anyone running into this issue as well:
I found the fix by renaming imuSubTopic from "imu" to "/imu", apperently "imu" does not always work.

closed 2016-08-22 by krijnen