PX4/PX4-Autopilot #5239: [MOCAP] when ATT_EXT_HDG_M is set to 0, the PX4 ignores all but the first MOCAP messages

AlexisTM, 2016-08-04:
Dear,

I am using the PixHawk (PX4, last stable version) with an OFFBOARD system to get his relative position. 

I publish on /mavros/mocap/pose topic.

Since I do not have the heading of the multicopter via my localisation system, I set the ATT_EXT_HDG_M to 0. 

Badly, when ATT_EXT_HDG_M is set to zero, only the **first** value is taken and the position stay steady afterwards. 

When ATT_EXT_HDG_M is set to 2, the Pixhawk see the correct (mocap) position, but with delays on his heading (read from PixHawk's IMU => Odroid XU4 => PixHawk. 

It looks like a bug but maybe it is intended to be so.

mhkabir, 2016-08-08:
We're phasing out INAV anyway, so could you please try LPE? Set the param `SYS_MC_EST_GROUP` to LPE + AttitudeQ.

AlexisTM, 2016-08-08:
I didn't knew that ! Thanks for the information ! 

I will soon try LPE then !

mhkabir, 2016-08-21:
@AlexisTM I believe this is resolved with LPE either way.

closed 2016-08-21 by mhkabir

AlexisTM, 2016-08-21:
It is not yet resolved because I am unable to make the flight right now.  (I will be able to test in September) Thanks for your support :D

AlexisTM, 2016-09-07:
It is resolved by using LPE, as you predicted. Thanks.