PX4/PX4-Autopilot #5240: Enable tighter C++ warnings per module, eventually as default

LorenzMeier, 2016-08-04:
@bkueng @dagar @jgoppert I would like to enable ```-Weffc++ -Wuninitialized``` for more modules and eventually as default. We should not embark on this pre-1.4 release, but for 1.5. I'm particular after uninitialised members, which I really feel we should enforce. Also other issues like pointer members should be handled (or the copy constructor forbidden).

dagar, 2016-10-24:
Some others that may help
 -Winit-self
 -Woverloaded-virtual
 -pedantic-errors

Disable copy constructors and assignment operator.

    NonCopyable & operator=(const NonCopyable&) = delete;
    NonCopyable(const NonCopyable&) = delete;
    NonCopyable() = default;

dagar, 2017-01-02:
-Wuninitialized is on. Closing in favor of #6082

closed 2017-01-02 by dagar