PX4/PX4-Autopilot #5243: Bug in baro driver in Snapdragon flight

langxm1223, 2016-08-05:
Hello @julianoes 
I tested several times about the baro sensor on the snapdragon, and found some potential bugs in the driver framework, which cause some abnormal outputs. [https://github.com/PX4/DriverFramework/blob/master/drivers/bmp280/BMP280.cpp](url)
I attached a possible fixing patch here. 
[BMP280.patch.zip](https://github.com/PX4/Firmware/files/403228/BMP280.patch.zip)
If you think it is correct, it could be merged to the main px4 branch.
Thanks.

LorenzMeier, 2016-08-05:
@julianoes This is the patch:

```diff
diff --git a/drivers/bmp280/BMP280.cpp b/drivers/bmp280/BMP280.cpp
index bbcd67a..b9e35b1 100644
--- a/drivers/bmp280/BMP280.cpp
+++ b/drivers/bmp280/BMP280.cpp
@@ -57,7 +57,7 @@ using namespace DriverFramework;
 
 // convertPressure must be called after convertTemperature
 // as convertTemperature sets m_sensor_data.t_fine
-int64_t BMP280::convertPressure(int64_t adc_P)
+uint32_t BMP280::convertPressure(int64_t adc_P)
 {
 	int64_t var1, var2, p;
 	var1 = ((int64_t) m_sensor_data.t_fine) - 128000;
@@ -78,7 +78,7 @@ int64_t BMP280::convertPressure(int64_t adc_P)
 	var2 = (((int64_t) m_sensor_calibration.dig_P8) * p) >> 19;
 	p = ((p + var1 + var2) >> 8) + (((int64_t) m_sensor_calibration.dig_P7) << 4);
 
-	return p;
+	return (uint32_t)p;
 }
 
 int32_t BMP280::convertTemperature(int32_t adc_T)
@@ -299,8 +299,8 @@ void BMP280::_measure(void)
 
 	m_synchronize.lock();
 
-	m_sensor_data.pressure_pa = convertPressure(pressure_from_sensor) / 256.0;
 	m_sensor_data.temperature_c = convertTemperature(temperature_from_sensor) / 100.0;
+	m_sensor_data.pressure_pa = convertPressure(pressure_from_sensor) / 256.0;
 	m_sensor_data.last_read_time_usec = DriverFramework::offsetTime();
 	m_sensor_data.read_counter++;
 
diff --git a/drivers/bmp280/BMP280.hpp b/drivers/bmp280/BMP280.hpp
index c6bc6a3..fdb3507 100644
--- a/drivers/bmp280/BMP280.hpp
+++ b/drivers/bmp280/BMP280.hpp
@@ -96,7 +96,7 @@ private:
 	// Q24.8 format (24 integer bits and 8 fractional bits)
 	// Output value of “24674867” represents
 	// 24674867/256 = 96386.2 Pa = 963.862 hPa
-	int64_t convertPressure(int64_t adc_pressure);
+	uint32_t convertPressure(int64_t adc_pressure);
 
 	// Returns temperature in DegC, resolution is 0.01 DegC
 	// Output value of “5123” equals 51.23 DegC
```

LorenzMeier, 2016-08-05:
@langxm1223 The patch only reduces the output range - can you confirm it actually fixes the issue in your testing?

langxm1223, 2016-08-06:
Hello @LorenzMeier , 
Thanks for your reply.
According to the bmp280 data sheet and the sample pressure and temperature conversion code, the temperature conversion must be done before the pressure conversion, as the latter one uses some result from the temperature conversion. This patch only fixed the issue that first some baro reading is wrong. However, during the test, I observed incorrect baro reading every once a while, after it is running for a while. I doubt there is still some issue in the driver framework bmp280 driver.
So I hope that someone could take a deep review of that.
Thank you.

LorenzMeier, 2016-08-06:
Ok, that makes sense and thanks for letting us know about the remaining issue.

closed 2016-12-24 by LorenzMeier