PX4/PX4-Autopilot #5244: Gazebo VTOL Not Working

tanenyi, 2016-08-05:
Already set up Gazebo, toolchains and other dependencies using the methods prescribed on the PX4 website.

The standard quadcopter build is working well but both the standard VTOL and the tail sitter VTOL loses control shortly after taking off although in GPS or mission mode. Wonder if it is a VTOL specific setting that should be changed?

[image]

[Video](http://gofile.me/6r9di/DisvqgeoW)

tumbili, 2016-08-05:
@tanenyi Can you set the parameter EKF2_MAG_TYPE to 1 and try again. We have an issue with 3d mag fusion but only in SITL.

jgoppert, 2016-08-05:
@tanenyi  It looks like the lift drag plugin is missing.
The vtol flight is part of the CI so it does work if you get the paths setup correctly. You might want to try:

LorenzMeier, 2016-08-05:
@tumbili I've disabled 3D mag fusion in SITL for now so that people don't run into it. Where is the associated issue for this?

jilinggui, 2016-08-05:
@jgoppert on my computer quadcopter, standard VTOL and the tail sitter VTOL build  work fine,but when i swtich to mission mode ,the qgc display "return, link loss",what happen? can you help me?

closed 2016-08-05 by LorenzMeier

tanenyi, 2016-08-08:
Apologies for delayed response.

@tumbili I set the parameter EKF2_MAG_TYPE to 1 and 2 (default is 0) but the same problem persists. I get a warning of **EKF GPS fusion timeout - resetting to GPS**, should I disable GPS also?

@jgoppert I used your branch to run SITL and while compiling, it includes the lift drag plugin you were mentioning but I still run into the same problem. There are 2 recurrent errors that are also present on the master build: **EKF baro hgt timeout - switching to GPS** and **EKF GPS hgt timeout - switching to Baro**. Would that have something to do with the IMU processing simulated data?

I get many warnings/info that the **simulator is slow. Delay added: xxxxx us.** but I think that is just my laptop being underpowered and should have little impact on SITL simulation.