PX4/PX4-Autopilot #5245: Pixracer: Sudden baro pressure and temperature jumps

mswingtra, 2016-08-05:
Yesterday in the flight we experienced the following weird behaviour:
>When flying in position controlled mode, the plane suddenly started descending between 1-2 metres.
After having a look at the logs we saw that the baro had a jump in its measurement of around 10 metres which caused the controller to descend (as it is filtered by the ekf it did not try to drop 10 metres instantly).
Changing pixracer and flashing older code version (one prior to the major changes in sensors structure) had the same problem.
The pattern is always the same. The baro temperature drops around 0.5°C then rises 1°C and goes back to normal. In the meantime, the baro altitude rises around 10 metres and then drops to normal again. The whole thing takes around 1.5 seconds. Though time wise it can also take longer.

Example:
[image]
[image]

Logs from two different pixracers:
http://logs.uaventure.com/view/eaUNJD4XABW8pxNqDmvWRZ
http://logs.uaventure.com/view/NAJGvkrZpUuuy4jHipZgkW

---

Afterwards, we started going through all the previous flights to see whether this has happened before.
We have spotted the same pattern on at least 6 different pixracer boards, flown on 4 different aircraft configurations (incl. wingwing) throughout last 3 months. It seem to happen at random points in time. I wonder if anyone else observed this issue.

Majority of the flights were we found this was with baro shielded the way you guys recommended - black foam tightly screwed:
[image]

Also, when baro measurements change due to lack of shielding, it looks much different:
[image]
[image]

We are now logging a long session on the bench to see how often it occurs and whether it depends on environment (though 4 different airframes prove this one wrong), so we'll update when we find out something new.

bkueng, 2016-08-05:
Could it be related to https://github.com/PX4/Firmware/issues/5150 ?

LorenzMeier, 2016-08-05:
@mswingtra Very likely a candidate issue. Unfortunately the original reporter didn't propose a fix and test - could you have a pass on that as soon as possible?

His last comment describes what to change: You need to implement the last operation from the datasheet correctly. Make sure to send the PR before you flight-test it so I can review the math change.

LorenzMeier, 2016-08-06:
I'm not sure this fixes it, but that's what the other issue discussed:
```
diff --git a/src/drivers/ms5611/ms5611.cpp b/src/drivers/ms5611/ms5611.cpp
index 300c0aa..4997a54 100644
--- a/src/drivers/ms5611/ms5611.cpp
+++ b/src/drivers/ms5611/ms5611.cpp
@@ -693,7 +693,7 @@ MS5611::collect()
        if (_measure_phase == 0) {
 
                /* temperature offset (in ADC units) */
-               int32_t dT = (int32_t)raw - ((int32_t)_prom.c5_reference_temp << 8);
+               int64_t dT = (int32_t)raw - ((int32_t)_prom.c5_reference_temp << 8);
 
                /* absolute temperature in centidegrees - note intermediate value is outside 32-bit range */
                _TEMP = 2000 + (int32_t)(((int64_t)dT * _prom.c6_temp_coeff_temp) >> 23);
```

CarlOlsson, 2016-08-06:
I did two flights with the wingwing
With current master c76c8fda044e91fb966504131239d83862831a0b
[image]
[image]

With dT declared as int64_t
[image]
[image]

I will upload the logs when I get back to wifi

CarlOlsson, 2016-08-06:
**Logs**
[master fw](http://logs.uaventure.com/view/GFgeVMHAgBLVRKQDhB2p2k)
[baro fix](http://logs.uaventure.com/view/VdsznwFtaJEiqsDkSHsXeB)

tumbili, 2016-08-08:
@CarlOlsson @bkueng Out of interest just wondering if anybody has looked at these logs yet?

LorenzMeier, 2016-08-08:
@CarlOlsson the log you shared does not include the same temperature range as the original one. Could you please clarify what aspects you were trying to illustrate with these logs? Can you repro a bench test with a hot air gun and the original temperature values exhibiting the issue?

CarlOlsson, 2016-08-09:
Hi, the logs from the wingwing was simply just to illustrate what changes in flight with or without the change of declaration. However we found a way to reproduce it to 100% yesterday. Michael will update here as soon as the postprocessing is done

mswingtra, 2016-08-09:
@LorenzMeier We have observed that issue yesterday, so we have applied the fix you proposed [couple of posts above](https://github.com/PX4/Firmware/issues/5245#issuecomment-238013321).
However, this has not helped and I am not really surprised because the fix is related to temperature compensation which, according to datasheet, should happen only for temperatures below 20°C.

Anyway, since it was very repetitive, we narrowed down what's triggering it. It has been happening every time we have sent change in parameters over telemetry. We suspect it has to do something with voltage levels across pixracer, so we will investigate it further with logic analyzer later today. 
Also, we have got RC spikes twice yesterday, and this has also seemed like being triggered by the same thing, just much less often - possibly RC receiver is not that susceptible as baro.

julianoes, 2016-08-09:
@LorenzMeier where is this baro fix? It looks to me that this potential fix would only matter in the case where temperature is below 20 deg C.

mswingtra, 2016-08-09:
We have done some checks, and long story short, with try and error method, we have found out that the issue has been fixed on PX4 master with commit 021f0840ae7f724bfdfd317dda3d57362017b835. Before that, issue is clearly present, even when no peripherals are connected (except USB and nutshell). Unfortunately for us, our last subtree update was 9 commits older, so we didn't have that fix. 

The thing here is that we do order our pixracers directly for AUAV, and we do not have any cheap chinese copies. So, could any of you guys check and confirm that this happens on your pixracers too? Just change any parameter and observe altitude in QGC - for the mentioned commit, and commit before it.

pkocmoud, 2016-08-09:
@mswingtra I would want to make sure that your issue is not related to https://github.com/PX4/Firmware/issues/5188 as you are interacting with both the baro and FRAM.

mswingtra, 2016-08-09:
@pkocmoud It is definitely related! We still observe RC spikes but it does not happen every time we send a param value to the pixracer. This is then quite similar to https://github.com/PX4/Firmware/issues/5170. We're using Taranis transmitter, and FrSky x4r receiver connected on S-BUS, so possibly different setup results in a different result.

Oddly though, Lorenz's commit fixes issue with baro.

mswingtra, 2016-08-10:
I am closing here, as the issue with baro jumps is solved.

The RC spikes are still present, but it seems to be related only to our custom changes, as we could not reproduce it with current master.

closed 2016-08-10 by mswingtra

nicolaerosia, 2017-05-29:
I'm also experiencing ms5611 sudden jumps in both temperature and pressure, I'm using Navio2 on RPI v3, master
```
10169561 ms5611 got crazy: -250.660004
48392360 ms5611 got crazy: -52.549999
53189758 ms5611 got crazy: -250.660004
73569122 ms5611 got crazy: -250.660004
81569288 ms5611 got crazy: -250.660004
165589487 ms5611 got crazy: -250.660004
174969255 ms5611 got crazy: -250.660004
183193163 ms5611 got crazy: -149.089996
198989181 ms5611 got crazy: -250.660004
220393282 ms5611 got crazy: -109.099998
247189379 ms5611 got crazy: -250.660004
285589511 ms5611 got crazy: -250.660004
292193399 ms5611 got crazy: -124.989998
304589347 ms5611 got crazy: -250.660004
```

nicolaerosia, 2017-05-29:
Taking a look at the raw values, it seems that if the temperature raw value is 0, the pressure value will also be broken, even more, if the pressure value read is zero, the next temperature reading will be broken (non-zero).