PX4/PX4-Autopilot #5247: LPE with LPE_GPS_ON=1: Sudden jumps at position hold

ecmnet, 2016-08-05:
With LPE defaults and LPE_GPS_ON during an outdoor flight, I've experienced sudden jumps due to GPS corrections (eps shown in red).  Position estimation is based on optical flow and GPS (17 Sats available). Shown below: EPV (red), LPOS.Y (Yellow) and SP.LPOS.Y (green)

[image]

Note: I'll share the log as soon I've sufficient upload bandwidth available

LorenzMeier, 2016-08-05:
That's an interesting problem to solve, but it's not really an LPE bug. It's merely at the core of GPS vision fusion and a new and important challenge to tackle.

fibr, 2016-08-06:
This exact type of thing crashed our car in DGC 2005. You can restrict (slew-rate) corrections to some of your states, which may work as a quick-fix. What I think is perhaps better is to have a tight control on max velocity and try to keep local info strictly in body frame / unpolluted with global coords. Not 'easy' either way.

LorenzMeier, 2016-08-06:
@fibr Thanks for the feedback! Would you feel running a small DKF with GPS only and fusing it on a delayed time horizon based on its internal covariance makes sense? When the fusion starts it would offset the GPS position to result in a zero update to the local position.

fibr, 2016-08-07:
TBH don't get your description. GPS is a KF already, unless you're dealing with raw pseudoranges. In principle I'd argue against creating further KFs, especially with interdependencies.  To try to clarify my comment above: as long as your local world is consistent and the local<->global mapping is controlled either with slewrate (easier) or appropriate jump logic (probably better), you should be alright.

Thinking about it now, an "easy" fix could be to artificially increase meas covariance of GPS updates not consistent with current state ("consistency/smoothness more important than truth"). Make sure you only do it on large enough jumps though, otherwise you'll mess up all your updates.

It's been a few years though, and this stuff is extremely implementation sensitive. Read my comments with plenty of salt :-)

LorenzMeier, 2016-12-26:
Stale

closed 2016-12-26 by LorenzMeier

saeidkazemi, 2017-02-12:
hi
We have some unknown problems to compile your code and we would be pleased if you could help us to fix it.
I have dealt with the problem of Position Control Mode actually the system rejects this mode.
We do not have GPS and want to run the Position Control with optical Flow Sensor.
We cannot find the LPE-GPS-on in Qground to turn it off, so how can we turn off the GPS?
We set the sys_group in LPE mode but again it rejects the position control mode. Do you know the reason?

Pixhawk version 2