PX4/PX4-Autopilot #5254: Logging for standard VTOL is broken in SITL

CarlOlsson, 2016-08-05:
@jgoppert 
After this commit 699b6a2cb340f4b6ec3979ae7748ff9c56cfdb89 standard_vtol is not producing any logs anymore in SITL.

LorenzMeier, 2016-08-05:
@CarlOlsson The log output location might have changed. @jgoppert Please advise.

CarlOlsson, 2016-08-05:
Oh you are right, maybe I was a bit to quick in my conclusion. Will give it another look

LorenzMeier, 2016-08-05:
The system does print the log location once logging starts. You should be able to figure it out from there.

jgoppert, 2016-08-05:
I can confirm this is working on my machine.

```bash
~/git/Firmware$ ls build_posix_sitl_default/tmp/rootfs/fs/microsd/
dataman  log  msgs_2016_08_05_17_01_49.txt
```

LorenzMeier, 2016-08-05:
@CarlOlsson I've reverted the logging / path change, so the path on master will be back to default. However, the file path will change again at a later point.

CarlOlsson, 2016-08-06:
Yes I can also confirm that it was working on my computer before you reverted the commit. Closing here

closed 2016-08-06 by CarlOlsson