PX4/PX4-Autopilot #5260: Pixracer: servos not active when _safety_off

kd0aij, 2016-08-05:
Not sure whether this is the intended behavior

kd0aij, 2016-08-05:
apparently this was due to PWM_DISARMED defaulting to zero for the standard AERT plane config.

LorenzMeier, 2016-08-06:
They actually should be active, only throttle should be locked in that state.

kd0aij, 2016-08-07:
@LorenzMeier I think parameters PWM_MIN/MAX/DISARMED should all be set in rc.fw_defaults. Currently only PWM_AUX variables are being set there, but 2101_fw_AERT just copies the PWM_xxx parameter values to bash variables of the same name. 

Was it intentional to use bash variables which shadow parameter values? That has been a source of confusion for me.

kd0aij, 2016-08-28:
closed pending further discussion

closed 2016-08-28 by kd0aij